arduino/arduino-ide #147: manually added Library does not compile

mrbbp, 2021-03-01:
**Describe the bug**
I use a library added manualy (https://github.com/cbm80amiga/Arduino_ST7735_STM)

**To Reproduce**
Steps to reproduce the behavior:
1. add the librairie in the libraries folder "libraries" in the sketch folder
2. Launch Pro IDE
3. Add the librairie from the menu (it's listed)
4. Compile the sketch

```
#define TFT_CS  PA1
#define TFT_DC  PA2
#define TFT_RST PA0
#include <Arduino_ST7735_STM.h>

#define SCR_WD 128
#define SCR_HT 160
Arduino_ST7735 lcd = Arduino_ST7735(TFT_DC, TFT_RST, TFT_CS);

```

**Desktop (please complete the following information):**
 - OS: MacOs 
 - Version: 10.14.6

ubidefeo, 2021-03-03:
@mrbbp 
what's the core/board you select to compile this?

thekunalsaini, 2021-03-03:
@mrbbp its happen because you are using beta version

ubidefeo, 2021-03-03:
@thekunalsaini can you be more specific on your comment?
I think it might be misleading and lacking context.

If you mean that this happens because @mrbbp is using the IDE Beta you might be wrong.
Every core which works with the Java IDE and the CLI is supposed to work with this IDE.

It is more useful to ask for contextual information than labelling things off with a quick answer.

thekunalsaini, 2021-03-03:
@ubidefeo  okay sir I understand, Sorry for any mislead

mrbbp, 2021-03-03:
i do not use the official stm32, but the Roger Clarks'one + use the Dan Drown package.json (to have the board in the Board Manager)...  http://dan.drown.org/stm32duino/package_STM32duino_index.json

reuse my old arduino sketch folder...with the librairies folder
and ohhhh, the lib is found.

today,I'm able to compile and upload with the IDE 2.0 beta 004 + STM32 board and a ST-LINK...

everything is good.

nethermind to add custom board,it needs package_json file, not in the IDE v1.