arduino/arduino-ide #150: Add advanced mode from Pro IDE

szerwi, 2021-03-01:
**Is your feature request related to a problem? Please describe.**
Arduino Pro IDE introduced many advanced features like Git integration or completly new file explorer, which gave possibility to open other files than .ino (like .cpp or .h). Unfortunately, the advanced mode is not available in Arduino IDE 2.0.0 beta-3

**Describe the solution you'd like**
Do you plan to import advanced mode features from Pro IDE to IDE 2.0?

ubidefeo, 2021-03-01:
@szerwi 
this new IDE comes from the Pro IDE, which gave us a lot of insight into what works, what's problematic and what users want.
We think that for some users the explorer can be confusing, but it doesn't mean advanced users don't need it.
As we move forward with an application that can satisfy the beginners and be appealing to the advanced, we will introduce more paradigms and useful features.

I'd say keep an eye open for new enhancements and new tools to come :)

In the meantime you can trigger the explorer view on using the advanced preferences.
Right now these are at the bottom left, and you can search for "toggle explorer view".
Mind that if that messes up your layout we do not support it and this view might be removed in favour of something more Arduino-friendly.

Hope you'll enjoy Arduino IDE 2.0 🙏🏼

szerwi, 2021-03-01:
@ubidefeo Thank you for the information. I also found there Git and other features from advanced mode ;) The only thing that I'm missing is closing tabs with .ino files. After enabling file explorer I can close tab with .h or .cpp file, but I can't close the tab with .ino file. Is that possible in IDE 2.0 the same as in Pro IDE, or not yet?

kittaakos, 2021-03-01:
> Unfortunately, the advanced mode is not available

You can access the _Explorer_ view from the _Command Palette_.
 - Press <kbd>F1</kbd> or <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>,
 - Use the _Toggle Explorer View_ command: just type `Toggle Explorer View` and hit <kbd>Enter</kbd> (or <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>E</kbd>).
[image]
[image]

For the time being, you can create any file from a terminal. Open a terminal with <kbd>Ctrl</kbd>+<kbd>\`</kbd> (or the _Open New Terminal (In Active Workspace)_ command) and create a file with any `filename` and `ext`ension:
```sh
echo > filename.ext
```
You should be able to open the file from the _Explorer_ view.
[image]

You can open the view for the Git integration with the _Toggle Source Control View_ command (or <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>G</kbd>).

I hope this helps during the transition.

kittaakos, 2021-03-01:
> Is that possible in IDE 2.0 the same as in Pro IDE, or not yet?

It's not possible at the moment.

szerwi, 2021-03-01:
@kittaakos Thank you for quick tutorial, I've already found that :)

I hope you won't abandon advanced features and you will keep upgrading them :)
Right now my sketch is built from multiple .ino files for better navigation and it is a bit annoying that I cannot close any of them, which was possible in Pro IDE.
[image]

ubidefeo, 2021-03-01:
@szerwi 
this IDE will not reintroduce things that have been removed in the transition from Pro IDE.
Like I said, these advanced hidden features might disappear in the future.
There will be other ways to get advanced features, and they will be announced in time :)

idea--list, 2021-03-02:
@ubidefeo 
`Like I said, these advanced hidden features might disappear in the future.`

I think this would be a fatal mistake. Even noobs are not dumb. Arduino project would only need to onboard poeple and telling them: hey until you do not feel comfortable, you should avoid the explorer and using any other files in a project than the starting .ino file (which is nothing just a renamed main.c). As soon as they explore the examples and try to write their own code they would really benefit from the explorer. I mean no one will just want to blink an LED after they explored the example. Chances are high they would want to mix blinking with some other functionalities. Why do not you just explain them in 1 tutorial that they could "externalize" each functionality to its own .h and .c files and just `#include` those in the .ino file as they need it.

It is really not as hard to explain all that even for a complete beginner that they could get their code organized and more easy to maintain (modify, debug, etc.) while telling them naturally they also can keep going on with a monolithic .ino file if they prefer.

Exactly the lack of doing anything else than editing an .ino file was the main reason i rather learned Mbed OS despite that has a much steeper learning curve. As boards released by Arduino do not contain just 8KB Flash anymore, sketches can contain more than 100 lines... and with the intention to remove "advanced feateres" you are just about to send Arduino IDE v2.0 back to the stone age where all arduino boards came with microscopic memory. In which case there was no point taking the effort to develop V2.0. As 1.8.13 is exactly what V2.0 is without the "advanced features".

szerwi, 2021-03-02:
@idea--list I totally agree with you. Arduino IDE should implement more and more advanced features, so there will be no need to migrate from Arduino IDE to VS Code + PlatformIO or any other alternative IDEs

ubidefeo, 2021-03-02:
@idea--list 
you can create `.c` and `.h` files in this new IDE, if that is your main concern.
We do it all the time, and it's very much needed by developers of libraries, so a better "explorer" view will come, we just want to make it our way.

The removal of an "advanced" switch doesn't mean we do not plan on catering advance users/features.
We just realised that exposing a whole set of features without giving it our touch wouldn't be different from just using VS Code.

I hope you'll want to wait for us to slowly introduce more advanced tools yet making them friendly.
I will not reveal our roadmap, but there' s a lot of stuff cooking and we want to make sure we gather feedback as we go.
Implementing all of it for the first public beta would have pushed this product release date to 2023 😬 

In the meantime, please, feel free to report bugs or malfunctioning into this repo.
I appreciate you taking the time to write detailed comments 🙌🏼 

ciao.ubi

Hutkikz, 2021-03-03:
I really liked the latest version of the Pro IDE enough that I switched to it full time.
Then yesterday I try the new 2.0 and was disappointed to find no advanced mode.  Boo.
Then I find a link to this thread telling how to get those features back.  Yeah!!
Only to also read you plan to remove this too.  **WHY???**

Not everything HAS to be different from VScode these features are now hidden enough from the newb's.
This feels like your teasing us with candy then intend to take it away. to be replaced (maybe) at some unspecified future time.

kevindawson, 2021-03-04:
@kittaakos 
Command Palette. ```F1``` Cool

- Toggle Explorer View -> ```Ctrl/⌘```+```Shift```+```E```
- Toggle Plugins View -> ```Ctrl/⌘```+```Shift```+```L```
- Toggle Outline View ->   _Not defined?_

Q, how do you enable them as default as there state is not retained, over an IDE restart

other wise this is cool ;)

kittaakos, 2021-03-04:
> * Toggle Outline View ->   _Not defined?_

<kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>I</kbd>

> as there state is not retained, over an IDE restart

The entire layout, including the opened views, is saved and restored between sessions. This works per sketch. If you open a new sketch, you start with the default layout. Currently, there is no way to change the default layout when opening a new sketch.

kevindawson, 2021-03-04:
@kittaakos 

I think you mean
```Ctrl/⌘```+```Shift```+```I``` -> Explorer view

no key-bindings
- Toggle Outline View -> Not defined?

kittaakos, 2021-03-04:
> I think you mean
> `Ctrl/⌘`+`Shift`+`I` -> Explorer view

No, I meant the _Outline_ view.

> no key-bindings
> 
> * Toggle Outline View -> Not defined?

You're right; there is no binding for it on Linux, and I guess you're on Linux 😊 Here is the Theia change that disabled the _Outline_ view keybinding on Linux: https://github.com/eclipse-theia/theia/commit/5c64e76e3ac6c47e733c751cbf705b55cd9785db

Looks like we have to adjust it for the new IDE.

kevindawson, 2021-03-04:
@kittaakos 

Thanks for the update, I look forward to seeing the implementation in future releases

So long and Thanks for all the Fish!

ubidefeo, 2021-03-04:
@kevindawson 
>Not EVERYTHING has to be different from VScode these features are now hidden enough from the newb's.
This feels like your teasing us with candy then intend to take it away. to be replaced (maybe) at some unspecified future time.

we collect this kind of feedback, and use it for future development.
I believe for advanced users having an explorer is a good thing, we just want to make it less hacky.
Right now we do not officially support the features which are not expose through the basic UI, and there's a reason for that.
Assume you use the advanced preferences and get your explorer view, then you get a bug in the UI and start posting issues here, we have something that will end up in our backlog because we have to prioritise things according to our roadmap.

What I see is that a long list of users are happy about the direction, as advanced/pro users ourselves we know what we need, and trust me when I say it's not far from what you need as a user 😄 

I humbly ask you to trust that we're doing the right thing.
When we announce a file explorer and it's totally disappointing to you I warmly invite you to voice your concern, and if necessary link to this issue and say "I told you so!".
The team reads every issue, every comment, every complain.
This is why we were able to bring out a Beta IDE which is overall pretty usable and pleasant.

I'm happy more advanced users post about these things, because they affect the roadmap.
We're a (very) small team, and we want our users to love the work we do 😉 

Looking forward to that "I told you so!", mate ✌🏼

kevindawson, 2021-03-04:
@ubidefeo 

Not my Quoit **mate**, waggle your finger else ware.

ubidefeo, 2021-03-05:
@kevindawson 
sorry, just hit enter on the suggestion box at the wrong moment.
thank you for pointing it out ;)

kevindawson, 2021-03-05:
@ubidefeo 

Cool

So long and Thanks for all the Fish!

runger1101001, 2021-05-02:
Hey - I'd just like to add my request to give the pro features some attention. I think you've done a wonderful upgrade to the core of the old IDE, and you've also successfully captured the old ease of use by recreating the old look and feel and menus to a large degree.
I think I fully understand the importance of this given the huge volume of documentation available for the old system. Continuity will keep all this knowledge alive.

But (IMHO) now it's time to do some new things that support your pro users. After all, they're the users who write all the libraries, board descriptions and other toolings that help to make Ardunio what it is.

With vscode, you've provided a technology core to support all kinds of new features easily, and more importantly, with many features already built in.

Here's some suggestions that would help me (keeping in mind I'm not a VSCode expert (yet) so obscure key combinations aren't that useful to me):
- Provide an advanced mode - seems to me this is a good way to keep the old IDE simple and like it was, but provide access to new IDE features. Why not provide a mode that just gives us what vscode can already do by default, rather than hiding it? Those features for me would be Version Control (Git), the Plugin Manager, the File Browser and the Console.
- Provide ways to override automatisms. Automatisms are great when they work. But when they don't you need a way to just set them explicitly. Specifically, I would like to override the automatic library detection and just include a library. And I would like to override the automatic board and port detection to just set values I want. Currently, the automatisms are a great hindrance when they don't work as expected.
- Provide convenient ways to work with libraries, most importantly [this one](https://github.com/arduino/arduino-cli/issues/1255) but also supporting symlinks to libraries, so we can easily work with different branches and versions of the same library in parallel. Maybe even something like a "New library..." feature would be cool?

PS: Before people point this out, I know I can use PlatformIO, but this comes with its own set of concepts and problems, it just isn't the same as Arduino IDE. If you're trying to create a library for Arduino and other Arduino users to use, there is no substitute to compiling and testing it with the Arduino IDE.

per1234, 2021-11-08:
>how do you enable them as default as there state is not retained, over an IDE restart

Dedicated issue for this here: https://github.com/arduino/arduino-ide/issues/538

wojciechsura, 2021-11-12:
In the comments you claim, that disabling advanced features is for the sake of beginner users. I think, that there are some questions, which may be worth answering.
* Having a lot of voices to restore the "advanced" functionality, how many voices there were to remove it in the first place?
* Did any of beginner users complained because he simply could open any file in the folder and close it? Given the whole Arduino project may be a single file, how would that make things harder for beginners?
* Why opening and closing arbitrary files is called "advanced" and hidden/removed from the application? This is a behavior people are already used to and removing it causes them to lack control over application, what is always frustrating?
* Why allowing user to have option to use Git IDE enhancements is something, that may be troublesome for beginner users? They doesn't really need to use it and in the interface this is merely an icon in the sidebar.
* Why don't you want to take advantage of users knowing their way around VS Code? It was designed also for beginners and it is very likely, that a lot of Arduino beginners will come with some experience in this IDE. They will then need to learn the new (forced) way of working again.
* Do you have any actual evidence supporting the fact, that the legacy mechanism (forced opening files in the project, disallowing opening files from subfolders, disallowing closing files) is actually more comfrotable for the users than having control over what is opened and what is not?
* Do you have any statistics, which shows, how many users of the Arduino IDE consider themselves beginner and how many - advanced?
* Why, having a very well crafted IDE for all users, would you want to *remove* "advanced" features instead of hiding them? Having "advanced" UI hidden does nothing bad for beginners, but removing "advanced" UI hurts advanced users.

The truth is that after understanding simple sketches like Blink and LCD demos, when you start thinking of more interesting projects, you will need a way of organizing your projects into structures and this is what VS Code is fantastic with. I actually wrote most of my projects in VS Code, using Arduino IDE only for compiling and burning binaries into boards. I moved to 2.0.0 beta in hope of having all VS Code goodness along with capability of working with boards directly only to learn, that you actually plan to remove all that functionality. This simply makes completely no sense and I'm curious, what stands behind such drastic and unnecessary decision.

ubidefeo, 2022-09-19:
hi everyone still here

As you may have learned, we have finally released IDE 2.0 to the public.
It's not something which will mostly have conversations here on GitHub, but the forums will populate with feedback, requests and so on.
The forum is where the basic users deliver feedback, besides our internal user testing issues.

Why am I writing on this thread again?
Because we have finally stabilised IDE 2.0 and we can give some love to more advanced features which we did not have time for a year or more ago.

Our development team is so small (and the community's PRs so few) that shifting focus can lead to a complete disaster, so someone (often me) has to take it into their hands to steer in one direction or the other.
I'm here also to take the blame when something doesn't make people happy.

We want to gradually bring more tools and options to our more advanced users, but remember that you always have the option to use VS Code and our fantastic @arduino/arduino-cli for the die-hard terminal addicts I also see myself belonging to :)

It's been a rocky ride, we won't be stopping here :v

idea--list, 2022-09-19:
@ubidefeo 
So, will you just please decide at last to bring back "advanced mode" (aka file explorer). I mean any one who has ever used any computers are familiar with file browsers no matter how we call it. People who never used computers before and thus may get confused by a file explorer, are not likely to start programming microcontrollers. ;-)

If your answer is no, then i arduino IDE 2.0 and al the hype about it is simply irrelevant for any user. By that i mean beginners will not notice any functional difference compared to 1.9, while also advanced users will rather keep their complex stacks instead of just installing only 1 software. So at the end nobody benefits from all the work the team has done for almost 2 years.

For sure you can take a decision as you wrote, you just seem to forget that also the users may take their decisions and exactly that is why so many users migrated to using stacks of VS Code + different tools. Actually the whole point why it was needed to rehaul the IDE.

per1234, 2022-09-20:
Thanks all. Since this is more of a general discussion than the actionable reports that are appropriate for this issue tracker, I'll go ahead and close it.

Please open dedicated issues to request any specific individual features you would like. Make sure to search the issue tracker for previous issues about that feature to avoid the creation of duplicates:

https://github.com/arduino/arduino-ide/issues?q=

closed 2022-09-20 by per1234