arduino/arduino-ide #151: Compilation error: Error: 2 UNKNOWN: exit status 1

TheAnton205, 2021-03-01:
**Describe the bug**
Any program I run and attempt to compile I get this error:

> c:/users/anton/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avr5/crtatmega328p.o:(.init9+0x0): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status
Compilation error: Error: 2 UNKNOWN: exit status 1

- I recently got the beta and no matter what I do I had this (yes I restarted my pc) and I even download the latest Git (2.30.1) 64-bit.

**To Reproduce**
Steps to reproduce the behavior:

1. Open any code
2. Click compile

**Expected behavior**
Expected for the code to compile.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 64-bit
 - Version: 2.0.0-beta3

**Additional context**
This could be due to from when some notification occurred which I couldn't figure out what it was for. I do remember that it led to [this link](https://code.visualstudio.com/docs/setup/linux#_visual-studio-code-is-unable-to-watch-for-file-changes-in-this-large-workspace-error-enospc). However, I don't see this notification anymore?

Pioxis, 2021-03-02:
I have the same error. I installed first DallasTemperature library from library manager, and for OneWire library I do the same.

After that I tried compile code at the bottom of the page.
But it ending with Error 2 UNKNOWN: exit status 1.
After that I installed manually libraries from .zip archives. (that works for me for old IDE). The same error.
```c
#include <DallasTemperature.h>
#include <LiquidCrystal.h>
#include <OneWire.h>

#define ONE_WIRE_BUS 2

OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature.
DallasTemperature sensors(&oneWire);

// Definicja wejść wyświetlacza ciekłokrystalicznego //definition of lcd output
LiquidCrystal lcd(4, 5, 6, 7, 8, 9);
void setup(void) {
  // Inicjacja czujnika //sensor initiation
  sensors.begin();
  // Inicjacja wyświetlacza //screen initiation
  lcd.begin(16, 2);
  lcd.clear();
}
void loop(void) {
  lcd.setCursor(0, 0);
  lcd.print("Temperatura:");
  lcd.setCursor(0, 1);
  sensors.requestTemperatures();
  lcd.print(sensors.getTempCByIndex(0));
  lcd.setCursor(6, 1);
  lcd.print("st.C");
  delay(1000);
} ```

TheAnton205, 2021-03-02:
> I have the same error. I installed first DallasTemperature library from library manager, and for OneWire library I do the same.
> 
> After that I tried compile code at the bottom of the page.
> But it ending with Error 2 UNKNOWN: exit status 1.
> After that I installed manually libraries from .zip archives. (that works for me for old IDE). The same error.
> 
> My OS is Ubuntu 20.10
> 
> ```c
> #include <DallasTemperature.h>
> #include <LiquidCrystal.h>
> #include <OneWire.h>
> 
> #define ONE_WIRE_BUS 2
> 
> OneWire oneWire(ONE_WIRE_BUS);
> 
> // Pass our oneWire reference to Dallas Temperature.
> DallasTemperature sensors(&oneWire);
> 
> // Definicja wejść wyświetlacza ciekłokrystalicznego //definition of lcd output
> LiquidCrystal lcd(4, 5, 6, 7, 8, 9);
> void setup(void) {
>   // Inicjacja czujnika //sensor initiation
>   sensors.begin();
>   // Inicjacja wyświetlacza //screen initiation
>   lcd.begin(16, 2);
>   lcd.clear();
> }
> void loop(void) {
>   lcd.setCursor(0, 0);
>   lcd.print("Temperatura:");
>   lcd.setCursor(0, 1);
>   sensors.requestTemperatures();
>   lcd.print(sensors.getTempCByIndex(0));
>   lcd.setCursor(6, 1);
>   lcd.print("st.C");
>   delay(1000);
> } ```
> ```

I see. I'm not sure if it is due to libraries, I still get this error even from examples without library (File>Examples>Basics>Blink) still causes this error yet there are no libraries?

Pioxis, 2021-03-02:
> I see. I'm not sure if it is due to libraries, I still get this error even from examples without library (File>Examples>Basics>Blink) still causes this error yet there are no libraries?

For Blink example from Basic folder of examples I have that error:
```Compilation error: Error: 2 UNKNOWN: no FQBN provided```

ubidefeo, 2021-03-03:
@TheAnton205 @Pioxis 
have you installed any cores?

which board have you selected?
If the platform for the selected board is not installed you should get prompted to install it.

please try to manually add support for your board via the boards manager on the left sidebar or from the "Tools > Board > Boards Manager" menu

Pioxis, 2021-03-03:
> @TheAnton205 @Pioxis
> have you installed any cores?
>
> which board have you selected?
> If the platform for the selected board is not installed you should get prompted to install it.
>
> please try to [...]

Additionally I get that info from my terminal, when Error appeared:
```daemon INFO inventory.Store.ConfigFileUsed() /home/pioxis/.arduino15/inventory.yaml
root ERROR Request compile failed with error: 2 UNKNOWN: no FQBN provided
```

When I first opened program I was prompted for select the board. I selected for Arduino UNO.
I have installed _Arduino AVR Boards by Arduino, version 1.8.3_

TheAnton205, 2021-03-04:
> @TheAnton205 @Pioxis
> have you installed any cores?
>
> which board have you selected?
> If the platform for the selected board is not installed you should get prompted to install it.
>
> please try to [...]

SOLVED:

For me I solved it by reinstalling the beta IDE and by reinstalling _Arduino AVR Boards by Arduino, version 1.8.3_

Everything seems working now, thanks.

ChuckBland, 2021-03-05:
Same issue here. A program that easily comiles and runs under the old IDE gets:

`collect2.exe: error: ld returned 1 exit status
Using library IRremote at version 3.0.3 in folder: C:\Users\ccbland\Documents\Arduino\libraries\IRremote 
Compilation error: Error: 2 UNKNOWN: exit status 1`

when compiled with V2. No other discernable complaints when using verbose mode on compile.

I did uninstall/reinstall IDE 2.0 to no avail.

Here's my program
```
#include "IRremote.h"

const int IR_RECEIVE_PIN = 11;
long keyPressed;

void setup()
{
    //setup serial monitor
    Serial.begin(115200);

    //initialize IR Receiver
    IrReceiver.begin(IR_RECEIVE_PIN, DISABLE_LED_FEEDBACK);
}

void loop()
{
    //check to see if a button was pressed
    if (IrReceiver.decode())
    {
        //a button press was pressed and decoded

        //get the code of the key that was pressed
        keyPressed = IrReceiver.decodedIRData.decodedRawData;

        //if that code is from the power button, tell us
        if (keyPressed==0xBA45FF00) Serial.println("Power Key Pressed");

        //wait for things to settle down
        delay(700);

        //resume receiving button presses
        IrReceiver.resume();
    }
}
```

bphermansson, 2021-04-17:
I think this is a case of bad error messages. I got the error too, "Compilation error: Error: 2 UNKNOWN: exit status 1" in the output window. I started the program from the console, and there I see:
```
daemon INFO /tmp/.arduinoIDE-unsaved2021316-1396774-1hhgvhh.m6o3/Blink/Blink.ino:1:10: fatal error: Servo.h: No such file or directory
daemon INFO  #include <Servo.h>
daemon INFO           ^~~~~~~~~
daemon INFO compilation terminated.
root ERROR Request compile failed with error: 2 UNKNOWN: exit status 1
```
If I install the servo lib everything is fine, so the error message in the output window is wrong.

ChuckBland, 2021-04-17:
Patrik,

I'd like to replicate what you've done. I'm not familiar with "starting
from the console". How is that done?

Chuck

bphermansson, 2021-04-18:
I started the program from the command prompt in Linux. You can du the same with Windows Powershell . You go to the program installation folder and run the program by entering it's name. Then you see all log messages directly.

ChuckBland, 2021-04-19:
Very cool interface.

Sadly I am no closer to solution. No complaints about Servo.h, which is
fine since I'm not including it.

Here's the errors when I compile.

daemon INFO inventory.Store.ConfigFileUsed()
C:\Users\ccbland\AppData\Local\Arduino15\inventory.yaml
daemon INFO inventory.Store.ConfigFileUsed()
C:\Users\ccbland\AppData\Local\Arduino15\inventory.yaml
daemon INFO Searching tools required for board arduino:avr:uno
daemon INFO Required tool [tool:
{ToolName:"arduinoOTA",ToolVersion:"1.3.0",ToolPackager:"arduino"}]
daemon INFO Required tool [tool:
{ToolName:"avr-gcc",ToolVersion:"7.3.0-atmel3.6.1-arduino7",ToolPackager:"arduino"}]
daemon INFO Required tool [tool:
{ToolName:"avrdude",ToolVersion:"6.3.0-arduino17",ToolPackager:"arduino"}]
daemon INFO Adding libraries dir [dir: C:\Program Files
(x86)\Arduino\libraries, location: ide]
daemon INFO Adding libraries dir [dir:
C:\Users\ccbland\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\libraries,
location: platform]
daemon INFO Adding libraries dir [dir:
C:\Users\ccbland\Documents\Arduino\libraries, location: user]
daemon INFO Resolving include IRremote.h for arch avr
daemon INFO   found better lib [lib: IRremote, prio: 5E9]
root ERROR Request compile failed with error: 2 UNKNOWN: exit status 1

thelwyn, 2021-05-26:
Same issue here. Win 10 Pro 64 bits. Arduino Uno, Core installed (as suggested by the IDE itself). Can't compile anything, same error message.

jerabaul29, 2021-06-18:
Same error here, opening a detailed issue about it now.

jeromegirard, 2021-07-17:
Same error here, but like @bphermansson, when I compiled with verbose, I saw the the error comes from a lacking library.
Add the library and everything is working fine.

d-i-ry, 2021-08-04:
upgrading to Arduino IDE v2.0.0 beta 9 worked for me

pibondchem, 2021-08-13:
I had this error specifically when I added

#include <LiquidCrystal.h>

in my code.

I'm using a Dell XPS, Windows 10

Arduino IDE
Version:2.0.0-beta.10
Date: 2021-08-05T08:23:32.577Z(1 week ago)
CLI Version:0.18.3 alpha [d710b642]

cirobruno, 2021-08-21:
Same error!

Out or the blue, I couldn't compile anymore! 

Now, as soon as I restart Arduino IDE, I get this message:
"Plugin runtime crashed unexpectedly, all plugins are not working, please reload the page. If it doesn't help, please check Theia server logs."
Wen trying to compile, "Compilation error: Error: 2 UNKNOWN: exit status 1".

So, I've update the board's extension to esp32 by Espressif Systems V1.0.6, getting
"Compilation error: Error: 2 UNKNOWN: exit status 1".

Then, after changing it to esp32 by Espressif Systems V1.0.5, the error became:
"bash: line 0: [: /Users/cirobruno/Documents/Arduino/Testes: binary operator expected
usage: cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file target_file
       cp [-R [-H | -L | -P]] [-fi | -n] [-apvXc] source_file ... target_directory
Compilation error: Error: 2 UNKNOWN: exit status 64".

Back to esp32 by Espressif Systems V1.0.4, same thing again!

What to do now?
Maybe I should migrate to VSCode...

markyousif, 2021-09-26:
im new to arduino and it says exit status 1 i dont know what to do

per1234, 2021-09-26:
@markyousif. This issue tracker is only to be used to submit detailed, formal bug reports and feature requests.

You need to ask for help on the Arduino Forum. I'm sure we'll be able to help you out over there:
https://forum.arduino.cc/

PR #664 (2021-12-06): Remove gRPC error code from error notifications
Compilation error before:
[image]

Upload error before:
[image]

Compilation error after: 
[image]

Upload error after:
[image]

Fixes #151.
closed 2021-12-06 by silvanocerza