arduino/arduino-ide #153: add accelerator/ shortcut keys for IDE menu on Windows

andrew-james-heard, 2021-03-01:
The V1 IDE had accelerator/ shortcut keys for IDE main menu. 
[image]

Please add these to the V2 IDE to improve keyboard accessibility.
[image]

tigoe, 2021-03-03:
+1 this

andrew-james-heard, 2021-03-03:
I've uninstalled for now/ remain with 1.8.13. There are no shortcut keys for editor code folding & no setting for using an external editor.

per1234, 2022-11-23:
Upstream issue: https://github.com/eclipse-theia/theia/issues/7063

wdoekes, 2022-12-13:
@per1234: can you add os:linux label as well? (Or remove the windows one.)

andrew-james-heard, 2024-03-18:
3 years later (since I last evaluated v2 IDE) & no progress. Even the Windows key F10, conventionally used to move focus to the main menu doesn't work.