arduino/arduino-ide #154: Upload to fast for old Nano and clones (Old Bootloader)

toogooda, 2021-03-02:
**Describe the bug**
Old IDE allowed you to select ATMega328p (Old Bootloader) which slowed down the upload speed to 57600. No such option so half my boards cannot be programed new system tries 115200 and times out.

**To Reproduce**
Steps to reproduce the behavior:
1. Use old aduino or clone 
2. upload blink

**Expected behavior**
Have to Old Bootloader option in boards select or change upload speed.

**Screenshots**
Have screenshots but don't see an option to insert or upload them.

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 10

**Additional context**
Used Aduino as ISP to burn bootloader using old IDE but it did not fix the issue

ubidefeo, 2021-03-02:
@toogooda 

I have personally tested the slow clones and they work.
You will only have to select the **old bootloader** option from the **Tools >> Processor** sub-menu

[image]

ubidefeo, 2021-03-02:
@toogooda  If that works for you, please close this issue 
thanks
u.

toogooda, 2021-03-02:
Yes that works thanks I used the board and port screen as I thought it had replaced the other menus.  @ubidefeo Should processor also be on the board and port screen screen?

closed 2021-03-02 by toogooda

ubidefeo, 2021-03-03:
@toogooda 
I was looking into that possibility, we have it on the Create Web Editor for targets with further options, but we haven't gotten to that level of granularity yet.
This version might require some workarounds at times, but we're confident we'll manage to fit more features and refine the ones it has over the next few months ✌🏼