arduino/arduino-ide #155: Serial monitor real time error

KerteszDomokos, 2021-03-02:
**The IDE 2.0 serial monitor has an error:**
*I have 2.0.0-beta.3 version, and windows 10.0.18363.959, and my laptop has Intel i5 proc, 8gb RAM, etc.*

The problem is this: my arduino (nano, old bootloader, rate: 9600bps) sent datas to pc, and the readed data has big delay displayed. I test the reading in PuTTY, and here arrive the datas real time.
The problem are biger in faster communication (test in 115200), and in slower communication are less (test in 1200).

I worked a project, and the old problem was the buffer save all data, but the printing or processing are slow.

<br>
My english isn't perfect sorry.

ubidefeo, 2021-03-02:
@KerteszDomokos 
can I ask you why you need to send such high throughput to the Serial Monitor since a human cannot possibly read it at those speeds?
sending data machine to machine is of course a whole other story.
I ask because people tend to put free running `Serial.print()` instructions into their loops and the receiving end often needs to buffer.
Just so you know, we will be reworking the whole serial pipeline at some point, because we are aware of this lag

KerteszDomokos, 2021-03-02:
Thanks the reworking. 
I use the monitor just supervision. 
I make a communication between a python and an arduino, and this datas I would supervision, but this is just debugging.

Thanks your help!

per1234, 2021-04-04:
Thanks for taking the time to submit an issue @KerteszDomokos .

I see we already have a prior issue report about this at https://github.com/arduino/arduino-ide/issues/79.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-04-04 by per1234

torruella, 2021-04-19:
The IDE 2.0 serial monitor has an error:
I have 2.0.0-beta.5 version, the serial monitor is always sending stranges bytes to the arduino board, it makes too difficult to send commands to the board, only with very slow speeds (under 4800 bds) i can do that.
any suggestions?
thanks in advanced.

KerteszDomokos, 2021-04-19:
> I see we already have a prior issue report about this at #79.

Please use the #79 Issue in this theme.

But what is the stranges bytes? Please send an example.

If you must faster communicatiun you can do it with Arduino ide 1.x.x version, with the [PuTTY](https://www.putty.org/), or other app.
The contributors already work in this problem (for example [here](https://github.com/arduino/arduino-ide/tree/serial-monitor-rate-limiter))

*Thanks*