arduino/arduino-ide #157: Install Button on already Installed Board

Cyb3rn0id, 2021-03-02:
"Install" Button appears even if a Board is already Installed. 
The writing "Installed" appears in the upper right corner of the box describing the board, when you go with the mouse over the box, the box, "INSTALL" button appears.
Maybe, in this case, is more appropriate showing "Re-Install"?
[image]

kittaakos, 2021-03-02:
> "Install" Button appears even if a Board is already Installed.

It's not a bug. This is how it works now in the Java IDE. You have installed the latest, `1.8.3` version, and you can install the last one: `1.8.2`:
[image]

> The writing "Installed" appears in the upper right corner of the box describing the board, when you go with the mouse over the box, "INSTALL" button appears.

Can you please explain this, @Cyb3rn0id? 👆

This is how it works:
 - if a platform is installed, you will see _INSTALLED_,
 - if you hover over, you will see _UNINSTALL_, you can click on it to uninstall the platform.
 
[image]

Of course, the UI/UX can be improved, but it's not an app malfunction.

Cyb3rn0id, 2021-03-02:
For me the "INSTALL" button in the lower-right corner maybe must say "ReInstall" when a platform is already installed or maybe this button must be disabled. But it's my opinion, maybe other people likes to read "Install" anyway.

Cyb3rn0id, 2021-03-02:
Sorry for the "bug" label, it's my own mistake. Obviously is not a bug, only an advice for the gui

ubidefeo, 2021-03-03:
@Cyb3rn0id 
Thank you for opening the issue so we could all look at it :)
Looking at this screenshot, the install button rightly allows you to install previous versions of the core, but the screenshot you sent me yesterday was showing `1.8.3`.
I just tested to see wether the GUI would not update at the installation moment, but cannot reproduce it.

@kittaakos just for your information, @Cyb3rn0id is one of my friends and sent me this screenshot yesterday which made me ask him to submit an issue.
As you can see in the screenshot I got from him yesterday it says "1.8.3 installed" and at the bottom "1.8.3 install"

[image]

ubidefeo, 2021-03-03:
In addition, that is not our core... and we can't figure out which one it is.
Maybe @per1234 knows it and he can confirm that this core is not fully compliant with the platform specs hence returning some weird versioning

kittaakos, 2021-03-03:
> In addition, that is not our core

@Cyb3rn0id, how did you install the _WiFi Duck Avr_ core? Did you add the 3rd-party URL via the _Settings_ UI and installed the core from the new IDE using the boards manager, or did you copy the core to the `directories.user/hardware` folder? Thank you!

> As you can see in the screenshot I got from him yesterday it says "1.8.3 installed" and at the bottom "1.8.3 install"

Yes. That's a bug. Do you know how I can reproduce it?

per1234, 2021-03-03:
>Maybe @per1234 knows it

Yeah, it's this one:
https://github.com/SpacehuhnTech/arduino
but that package index only has "WiFi Duck AVR Boards" versions 1.8.1, 1.8.11, 1.8.12, 1.8.13. No 1.8.3. I do see that there is a 1.8.3 tag in the platform repo itself:
https://github.com/SpacehuhnTech/arduino-avr/tree/wifiduck-avr-1.8.3

So I don't know what @Cyb3rn0id  has installed. However, I can reproduce this. It happens when there is only one platform in the package index. Here's the URL for one such index:
https://rawgit.com/hunianhang/nufront_arduino_json/master/package_tl7788_index.json
[image]

kittaakos, 2021-03-03:
> So I don't know what @Cyb3rn0id has installed. However, I can reproduce this. It happens when there is only one platform in the package index. Here's the URL for one such index:
> https://rawgit.com/hunianhang/nufront_arduino_json/master/package_tl7788_index.json

Thank you!

per1234, 2021-03-03:
>It happens when there is only one platform in the package index.

Reading it again, I don't think I explained it very well. Even though I'm sure Akos was able to decipher my meaning, I'll clarify to make sure there's not confusion for any other interested parties. What I meant to say is that it happens when the package index only contains a single release of a platform. I'm certain the same would occur even if the package had multiple different platforms (e.g., WiFi Duck AVR Boards, WiFiDuck SAMD Boards).

per1234, 2022-03-06:
Closing in favor of the duplicate https://github.com/arduino/arduino-ide/issues/781

Even though this one came long before https://github.com/arduino/arduino-ide/issues/781 (my bad), I think https://github.com/arduino/arduino-ide/issues/781 makes the issue a little more clear. I'll add a comment there to make sure the valuable discussion here is still available.

closed 2022-03-06 by per1234