arduino/arduino-ide #158: Arduino IDE 2.0 don't init

jcamplifiers, 2021-03-02:
**Describe the bug**,
Arduino IDE don't start if  Shure System API ShureDeviceApiService is running.

 
**Desktop (please complete the following information):**
 - OS: Windows
 - Version:  10 20H2

**Additional context**
If Shure Designer is installed on your system, port 50051 is used so the Arduino IDE 2.0 does not just start and stays in the about window.

2021-03-03 00:00:00 daemon INFO Restarting daemon in 4 seconds...
2021-03-03 00:00:01 daemon INFO Restarting daemon in 3 seconds...
2021-03-03 00:00:02 daemon INFO Restarting daemon in 2 seconds...
2021-03-03 00:00:03 daemon INFO Restarting daemon in 1 seconds...
2021-03-03 00:00:04 daemon INFO Restarting daemon now...
2021-03-03 00:00:04 daemon INFO Starting daemon from C:\Users\jose\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
2021-03-03 00:00:04 daemon INFO Using config file: c:\Users\jose\.arduinoIDE\arduino-cli.yaml
2021-03-03 00:00:04 daemon INFO arduino-cli.exe version 0.16.1
2021-03-03 00:00:04 daemon INFO Setting up Prometheus metrics on :9090/metrics
2021-03-03 00:00:04 daemon INFO Starting daemon on TCP address 127.0.0.1:50051
2021-03-03 00:00:04 daemon INFO Failed to listen on TCP port: 50051. Unexpected error: listen tcp 127.0.0.1:50051: bind: Intento de acceso a un socket no permitido por sus permisos de acceso.
2021-03-03 00:00:04 daemon INFO Failed to start the daemon.
2021-03-03 00:00:04 daemon ERROR Error: Failed to listen on TCP port: 50051. Unexpected error: listen tcp 127.0.0.1:50051: bind: Intento de acceso a un socket no permitido por sus permisos de acceso.
    at Socket.<anonymous> (C:\Users\jose\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\arduino-ide-extension\lib\node\arduino-daemon-impl.js:307:58)
    at Socket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:181:23)

kittaakos, 2021-03-03:
Thanks for reporting it, @jcamplifiers.

> Intento de acceso a un socket no permitido por sus permisos de acceso.

I am not sure what the error message means, but you can adjust the port in the CLI config. The CLI config is under `C:\Users\your_user_name\.arduinoIDE\arduino-cli-config.yaml` on Windows. Stop the app, open this file with your favorite text editor, and change the default `50051` port to something else. Let us know how it worked. Thanks!

jcamplifiers, 2021-03-03:
Works! Thanks!

closed 2021-03-03 by jcamplifiers