arduino/arduino-ide #159: "Serial" is not recognized and is considered as an "Undeclared Identifier"

TheUnityGuru, 2021-03-03:
**Describe the bug**
The class "Serial" is not recognized and is considered a "Undeclared Identifier"

**To Reproduce**
Just try this to initialize the hardware serial in the void setup.

**Expected behavior**
Well, nothing is supposed to happen. The IDE is supposed to recognize it

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.0.0

**Additional context**
Just downloaded it, installed the arduino AVR boards, and tried out this simple sketch. NOTHING ELSE HAS BEEN DONE

matios13, 2021-06-24:
I have the same issue, Windows 10 , Arduino  nano RP2040 connect 
, is there any walkaround ?

Wolfy7, 2021-08-12:
Same issue here. Windows 10 Pro 21H1 (19043.1165).

[image]

Version: 2.0.0-beta.10
Date: 2021-08-05T08:23:32.577Z
CLI Version: 0.18.3 alpha [d710b642]

doronbehar, 2023-02-04:
It seems that whenever a line such as:

```cpp
extern Serial_ Serial;
```

is used in Arduino's code ([`USBAPI.h` in this case](https://github.com/arduino/ArduinoCore-avr/blob/24e6edd475c287cdafee0a4db2eb98927ce3cf58/cores/arduino/USBAPI.h#L154)), all methods inside `Serial_` are not detected as belonging to the `Serial` class actually used in the users' sketches. I experience the same issue with the `Keyboard.h` external library. See:

https://github.com/arduino-libraries/Keyboard/blob/25ab3cdfd2d2cf93f2315fd4c13ef22c9622ba50/src/Keyboard.h#L146

Here's how [I experience this with Neovim](https://neovim.discourse.group/t/arduino-language-server-libraries/3668):

[image]

Lastaapps, 2024-02-01:
If someone ends up here, a simple workaround is to define the `USBCON` macro, either at the beginning of `main.h`, or in your 
clangd config `.clangd` like this:

```yaml
CompileFlags:
  Add: [
    -DARDUINO=186, # To prevent libs from using WProgram. (not needed here)
    -DUSBCON=,     # To prevent the Serial not being defined
  ]
```
Make sure this does not overwrite the normal macro value during the compilation process.