arduino/arduino-ide #162: Error message when no board selected is bad for beginners

tigoe, 2021-03-03:
When you try tp compile  with no board selected, you get the message 

```
Compilation error: Error: 2 UNKNOWN: no FQBN provided
```

This is poorly written. Beginners, and even many advanced users, will not know what a FQBN is. Rewrite this in plan language,  e.g. 

```
You need to select a board type before compiling
```

Steps to reproduce:
1. Open a new sketch without selecting a board
2. Compile

thekunalsaini, 2021-03-03:
@tigoe for understanding, for you kindly provide screenshot

ubidefeo, 2021-03-03:
@tigoe we're already looking into how to make this better.
Issue #37 covers this and more. The **low** priority label was assigned because we were in release mode.
It will soon get back to **medium** because it's source of confusion.

PS: we're all very happy to see you use it <3

per1234, 2021-03-03:
Closing as duplicate of #37 

@thekunalsaini please refrain from commenting unless you have directly relevant information to share.

closed 2021-03-03 by per1234