arduino/arduino-ide #163: An unexpected error occurred when locating the Git executable

seaxwi, 2021-03-03:
**Describe the bug**
Every time I launch the IDE on Windows I get this error.

**To Reproduce**
Steps to reproduce the behavior:
1. Close the IDE if open
2. Open the IDE
4. See error

**Expected behavior**
For the error not to appear, as it doesn't seem to affect IDE functionality.

**Screenshot**
[image]

**Desktop (please complete the following information):**
 - OS: [Windows 10 Home 20H2 19042.804]
 - Version: [2.0.0-beta3]

thekunalsaini, 2021-03-03:
@seaxwi I have Tested this issue no it's completely fine.
Please check your system, maybe it's because of your system problem.
If you have any urgent work, please download any stable release at https://www.arduino.cc/en/software#experimental-software

thekunalsaini, 2021-03-03:
I hope your doubt is cleared @seaxwi

per1234, 2021-03-03:
Thanks @seaxwi!

In case it provides any extra useful information for the investigation, here's the terminal output from when this happens:
```
root INFO Git askpass helper is listening on http://127.0.0.1:53693.
root ERROR Error: Git installation not found.
    at C:\Users\per\Downloads\arduino-ide_2.0.0-beta.3_Windows_64bit\arduino-ide_2.0.0-beta.3_Windows_64bit\resources\app\node_modules\find-git-exec\lib\find-git-exec.js:69:69
```

ubidefeo, 2021-03-16:
@seaxwi 
is any of your sketches (or your sketchbook) a Git repository?
I think Theia might find the `.git` folder and consider the folder a Git repo,
hence looking for ways to handle it and missing the `git` executable in the path.

Let me know if that is the case, but we should investigate further.
@per1234 do you have time to give it a shot by making sure `git` is not known to your system and `git init` a sketch?
Of course if you have sketches as Git repos then my guess is wrong

per1234, 2021-03-16:
This occurs even when there are no sketches, libraries, or boards platforms installed.

There have been multiple other reports of this causing confusion on the forum.

seaxwi, 2021-03-19:
> is any of your sketches (or your sketchbook) a Git repository?

@ubidefeo No.

I'm also able to replicate this issue on a recent Windows 10 install which has not been configured with git at all.

ubidefeo, 2021-03-20:
@seaxwi 
I have added high priority to this issue because we have a few reports and need to investigate deeply what the cause of the issue can be.
Hope to have a fix soon, at least in a Nightly Build

BCGuy, 2021-03-22:
The windows version that also had problems was in my case 10.0.19042 Build 19042

kittaakos, 2021-04-01:
Closed via https://github.com/arduino/arduino-ide/pull/294.

closed 2021-04-01 by kittaakos