arduino/arduino-ide #164: Com port busy

BitsAndDroids, 2021-03-03:
**Describe the bug**
Can't upload to the device when I've uploaded the code once. It seems like I'm able to write to a com port once and after that the connection will always be busy. I have to physically switch ports for the IDE to recognize that the port isn't busy. If I use the old IDE I can upload without any issues (while the 2.0 still gives the busy error).

**To Reproduce**
Steps to reproduce the behavior:
1. hit Upload
2. Hit upload / serial monitor
3. Port is always busy

**Expected behavior**
Port to be available

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
Add any other context about the problem here.

elsatch, 2021-03-03:
I found this very exact problem yesterday and could not solve it, by plugin, unplugin, reseting the board, etc.

I am also using Windows 10 and I was using two boards at the same time. The problem only happened in one of the boards.

I could upload to an Arduino Uno board without any problem several times.
I found the error of serial port busy when uploading to an Arduino Nano 33 BLE Sense.

Maybe this error happens only with certain boards/cores?

[image]

alranel, 2021-03-08:
Hey @elsatch! 
Not sure this is your case, but I noticed this happens whenever you have another sketch window with the serial monitor open. In this case the IDE should be smart enough to close the serial monitor in any other open window if it is attached to the same serial port you're uploading to.

alranel, 2021-03-20:
I noticed that the serial port stays busy even after you close serial monitor.

1. Start the IDE from scratch
2. Upload a sketch that is blocking on Serial (`while (!Serial) {}`)
3. Note that your sketch is actually waiting for serial port
4. Open serial monitor: sketch runs correctly
5. Close serial monitor
6. Reset the board
7. Note that your sketch is running, which means serial port is still OPEN despite serial monitor is hidden
8. Open another sketch in a new window
9. Try to flash that sketch
10. Note that it will say that serial port is busy (because it's actually busy in the first window)
11. Close the first window
12. Try to flash the sketch
13. Note that it will actually upload

per1234, 2021-11-27:
Hi @BitsAndDroids and @elsatch. We are still waiting for you to provide the information requested in the previous reply so that we can proceed with the investigation or evaluation of this.

BitsAndDroids, 2021-11-27:
@per1234 My apologies for the late reply. As of the latest release I haven't encountered this issue. For me, it is completely fixed.

closed 2021-11-27 by per1234