arduino/arduino-ide #167: Unable to select content from serial monitor

eclipse1985, 2021-03-03:
**Describe the bug**
I want to perform a copy/paste operation from serial monitor output, but the selection disappear on the mouse up

**To Reproduce**
Steps to reproduce the behavior:
1. Print something on the serial monitor
2. Try to select some text

**Expected behavior**
Selection does not disappear

**Screenshots**
https://user-images.githubusercontent.com/6736700/109802275-e1db2d00-7c1f-11eb-8e38-022a2275b8ed.mov

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: Mojave 10.14.6 (18G8012)

**Additional context**
Even with auto-scroll on and off

kittaakos, 2021-03-03:
Thank you for reporting it, @eclipse1985. The copy/paste is tracked [here](https://github.com/arduino/arduino-ide/issues/105#issue-818838115):

> * We get copy/paste/find/select-all for the _Serial Monitor_ for free.

I am closing your issue as a duplicate.

closed 2021-03-03 by kittaakos

per1234, 2021-12-31:
This bug is now being tracked at https://github.com/arduino/arduino-ide/issues/247