arduino/arduino-ide #169: File Menu is cropped both in horizontal and vertical directions

elsatch, 2021-03-03:
**Describe the bug**
File menu seems to be cropped: menu items are not shown complete plus, the menu shows a scroll to discover new items. The resolution is high enough to show all items without problems, but it is not doing it.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Arduino IDE 2.0
2. Open Menu, File
3. Titles are not displayed properly

**Expected behavior**
File menu should show all items uncropped and without needing to scroll it

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.0.0-beta3
 - Screen resolution: 1920x1080

**Additional context**
Add any other context about the problem here.

kittaakos, 2021-03-03:
What is your _Scale and Layout_ Windows setting under _Display_, @elsatch? I set my display to 1920x1080, tried with 150% and 175% scales both with maximized and normal window; I could not reproduce the defect.

elsatch, 2021-03-03:
My screen is setup to 1920x1080 and 150% scale. I am using a Wacom ONE display (but I think that wouldn't be the root cause). Error happens both maximized and non maximized sizes in this display.

This is an auxiliary monitor. If I move the Arduino window to other monitors, set at 100 and 125% scales, the file menu works  as usual, without problems.

Find attached a screenshot of my screen config for this particular display: 
[image]

kittaakos, 2021-03-03:
> This is an auxiliary monitor. If I move the Arduino window to other monitors, set at 100 and 125% scales, the file menu works as usual, without problems.

Thanks a lot for the details 👍 It's beneficial.