arduino/arduino-ide #173: Verify Button does not change colour when activated.

BCGuy, 2021-03-04:
Verify Button does not change colour when activated, yet.

ubidefeo, 2021-03-04:
@BCGuy none of the buttons have rollover or press yet.
Thank you for the suggestion

ubidefeo, 2021-03-19:
Closed by #190 
@BCGuy , please check the behaviour in Beta 4 and close if solved

BCGuy, 2021-03-19:
sorry i am so new to this, i did not know i was responsible to close it

great stuff you guys do!

ubidefeo, 2021-03-19:
no worries
just confirm it works :)

BCGuy, 2021-03-19:
When I install the beta 4 version onto my windows machine, after it opens
it also presents the message "an unexpected error occurred when locating
the Git executable and it will not compile and gives the compile window
message "Compilation error: Error: 2 UNKNOWN: no FQBN provided"

but the colour changes on the verify button

BCGuy, 2021-03-19:
And I learned what I say in emails appears here.  Cheers.

ubidefeo, 2021-03-20:
Thanks @BCGuy 
The `git` error you reported is already being tracked by us in #163 
We're trying to figure out why it only happens to some users.
Would you be so kind to comment on #163 and add your Windows version for reference?

Thanks
u

kittaakos, 2021-04-01:
It still does not work for me. I can see the yellow background for verifying, but not for uploading.
[image]

I am using
```
Version: 2.0.0-beta.4-nightly.20210401
Date: 2021-04-01T03:09:59.931Z
CLI Version: nightly-20210329 alpha [ac39429e]
```

CC: @fstasi

ubidefeo, 2021-04-01:
same here, @kittaakos

per1234, 2022-01-12:
Related: https://github.com/arduino/arduino-ide/issues/279

per1234, 2022-08-03:
Closing as fixed by https://github.com/arduino/arduino-ide/commit/5ff9ce0028726a01d07511897150d07541a7b476 / https://github.com/arduino/arduino-ide/pull/1194

PR #190 (2021-03-10): [Toolbar] Hover style and Toggle State [ATL-1107][ATL-1045]
## Why
- We are missing hover and pressed state on the buttons, which can be confusing to users switching from Java IDE.
- The Java IDE has the buttons remain “activated” until an operation is carried out.
- We want the "Verify" and "Upload" buttons (and menu items) to be disabled while the corresponding operation is in progress.

## How
- added "hover" behavior to all toolbar items, via CSS rule
- added "toggle" state to arduino toolbar buttons, by adding a prop, binded to the relative registry command. The `isToggled` function of th [... cut]
commit 1a308b49 (2022-07-18): busy state for upload sketch button (fixes #173)
closed 2022-08-03 by per1234