arduino/arduino-ide #176: Projects Tab inside the IDE.

Beezey, 2021-03-04:
Opening new .ino file opens a new window. Sometimes it feels exhausting and confused working with multiple windows.
A project managing tab would be much appreciated for opening multiple projects in one window.

ubidefeo, 2021-03-05:
@Beezey 
Everyone has different styles of being organised.
Although there are no plans to introduce such feature, we'll leave this here and if it gets enough 👍🏼 we'll consider it.

I personally would find this more of a UI cluttering element compared to having a single sketch/project and all its part in its own windows and once it's closed I'm done with it :)

This is also the behaviour in the Java IDE which we have tried to maintain in order to ease transition.
At some point in IDE Pro's development I remember being able to have multiple sketches in the same window but the mash of tabs related to one or another cause more trouble than benefits.
A Project Tab at the top or side will definitely take GUI estate which we should try and avoid.

Definitely an interesting point, let's see how the community reacts to the suggestion
u.

Karl255, 2021-03-07:
@Beezey
There's actually already a feature that allows exactly that. For that you first need to unhide the explorer view with Ctrl + Shift + E or form the command palette (F1) > Toggle explorer view. Then right-click inside the explorer view or click the button with the 3 horizontal dots at the top of it and then click on "Add folder to workspace". Each folder then corresponds to its own project.