arduino/arduino-ide #177: Better status filtering in library and board manager, especially for updates

ckuethe, 2021-03-05:
**Is your feature request related to a problem? Please describe.**
Nope, doesn't seem like an existing open problem

**Describe the solution you'd like**
I would like to see something similar to the status dropdown in the old IDE to select board/libraries with available updates.

I tried typing `installed`, `type:installed`, and `status:installed` in the search field and it didn't show me only my installed libraries instead of all the libraries available. Nor did `updatable` and `upgradable` show me those which had available updates. Finally, I didn't find a way to force a check for updates or to see when updates were last checked.

**Describe alternatives you've considered**
As mentioned above, tried a few obvious guesses in the search bar to filter on some criteria. Also tried right-clicking on the library and board manager icons but those also didn't trigger any filtering options.

**Additional context**
Looks like someone else wants this feature too (https://forum.arduino.cc/index.php?topic=730437.0) but I didn't see it when I looked through the issues here, so I figured I'd open an issue.

ubidefeo, 2021-03-05:
thank you @ckuethe 
this is definitely one of the missing features.
We are going to release a better search for the next Beta but filtering is not in there yet.

It'll come :)

ckuethe, 2021-03-05:
Thank you! I'm really enjoying the new IDE and I appreciate all the work that has gone into it so far.

CraneZ-dev, 2021-04-18:
Old IDE has reminder/notification about available boards/libraries updates and prebuild filter for the boards and libraries with available updates. It is more useful and convenient way, than typing the filter criteria again and again every time your remember that you need to check for updates yourself manually.

tfinks, 2021-08-03:
Just downloaded IDE 2.0.0-beta.9. There is apparently not yet a Library Manager feature providing selection of ALL, UPDATABLE, INSTALLED, ARDUINO,... . This selection feature in the previous Library Manager is needed in the new IDE IMHO.

mklemarczyk, 2022-03-06:
Still not present, it is very needed feature. I can not find what is installed, and what is not.

Xylopyrographer, 2022-04-16:
A big project and the effort is appreciated.
I'm "plus one-ing" the request(s) that the 2.0 Library Manager support all the functionality of the 1.8 Library Manager. 
Priority would be:
1. Filters for "Installed", "Updatable". Very high priority.
2. Toggle to show only the Library title in the list
3. Option to pick & install a specific library version
4. Install .zip libraries
5. Notifications of available updates to installed libraries and installed boards

Comments based on 2.0.0-rc5-nightly-20220416

Durobot, 2022-05-27:
Please consider implementing at least filters for installed and updatable libraries.
Still can't tell which libraries can be updated in
Version: 2.0.0-rc6
Date: 2022-04-20T10:22:38.001Z
CLI Version: 0.21.0 [10107d24]

kittaakos, 2022-08-28:
Can somebody please help with what the following types mean in the IDE (1.x):

_Library Manager_:

[image]

_Boards Manager_:
[image]

More formally, how to create a predicate for these 👆 types when the object is a [`LibraryRelease`](https://github.com/arduino/arduino-cli/blob/aa41d72ee6fd422980a2ba43310c269bf0ae2e1f/rpc/cc/arduino/cli/commands/v1/lib.proto#L158-L188). I looked into the [source of the IDE (1.x)](https://github.com/arduino/arduino/) but could not find any meaningful with `grep`.

Thank you!

kittaakos, 2022-08-28:
> I'm "plus one-ing"

> 2\. Toggle to show only the Library title in the list

What is this?

per1234, 2022-08-28:
Hi @kittaakos 

>create a predicate for these 👆 types when the object is a [LibraryRelease](https://github.com/arduino/arduino-cli/blob/aa41d72ee6fd422980a2ba43310c269bf0ae2e1f/rpc/cc/arduino/cli/commands/v1/lib.proto#L158-L188)

They are provided via the `types` field:

https://github.com/arduino/arduino-cli/blob/aa41d72ee6fd422980a2ba43310c269bf0ae2e1f/rpc/cc/arduino/cli/commands/v1/lib.proto#L178

As for the platforms, it seems the only place this information is provided is here:

https://github.com/arduino/arduino-cli/blob/aa41d72ee6fd422980a2ba43310c269bf0ae2e1f/rpc/cc/arduino/cli/commands/v1/board.proto#L95

which is probably not useful in the context of Boards Manager. I guess Arduino CLI needs to provided in [the `cc.arduino.cli.commands.v1.Platform` message](https://arduino.github.io/arduino-cli/dev/rpc/commands/#cc.arduino.cli.commands.v1.Platform). It looks like there is a poor design of these interfaces, where each is somewhat redundant to the other, while each also has unique deficiencies.

per1234, 2022-08-28:
>>2. Toggle to show only the Library title in the list
>
>What is this?

A "toggle" (e.g., a checkbox) would cause the additional information (e.g., `sentence`, `paragraph` values) to be shown or hidden for the libraries in Library Manager, reducing the verbosity of the listings.

Despite what is implied by @Xylopyrographer's comment, Arduino IDE 1.x does not have any such capability.

Hopefully @Xylopyrographer will provide clarification, but I think the use case would be where you know exactly the library name you want to install, but a search for that exact name (e.g., `servo`) still returns a lot of results. Reduced verbosity would mean less scrolling to reach the library you are looking for. IMO, the better way to provide a better user experience in this use case would be the enhanced search syntax I proposed at https://github.com/arduino/arduino-cli/issues/1535

commit 5cb9166c (2022-08-31): Implemented filter and update all for libs/boards.

Closes #177
Closes #1188
closed 2022-08-31 by kittaakos