arduino/arduino-ide #178: Additional board Manager URLs (preference) - don't download the json file

mrbbp, 2021-03-05:
**Describe the bug**
When you add a package_xxx_index.json in the field in the preference, it never load the file, even after a quit/relaunch app

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Preference'
2. Click in the field of Additional....
3. add an url

**Expected behavior**
the file should be downloaded to appear in the Board Manager

**Desktop (please complete the following information):**
 - OS: MacOs
 - Version: 10.14.6

**Additional context**
if the file manually added to the Arduino15 folder(on Macos) the board is here.
In fact it does not download the file

ubidefeo, 2021-03-06:
@mrbbp 
which particular URL are you trying?
I have just installed `stm32duino` and it worked

korbin07, 2021-03-06:
I tried "http://arduino.esp8266.com/stable/package_esp8266com_index.json"
and did not work but it does work on 1.8.13.

ubidefeo, 2021-03-07:
@mrbbp 
I'm not able to replicate this one.
It worked out of the box for me, and it's actually one of the packages we've been testing with because we know how large the user base is

https://user-images.githubusercontent.com/1894487/110231409-e8b3b980-7f17-11eb-940e-1d4d8da01c1e.mp4

mrbbp, 2021-03-08:
ok sorry for my late answer...
i've tried this one
https://raw.githubusercontent.com/mrbbp/STM32Duino_package/main/package_monTestLocal_index.json

it's a test purpose just to try that my json is good.

When I rebooted my mac .. by relaunching IDE2.0, a message asked me to install my file ... so he finally saw it, but only after a machine reboot .... Goddamn I'm on MacOS not on Windows, you don't restart your Mac every morning to manage boards in Arduino... :D

closed 2021-03-08 by mrbbp

ubidefeo, 2021-03-09:
thank you for the update, @mrbbp