arduino/arduino-ide #181: In preferences window, the pre-selected theme is different than active theme.

Mammad900, 2021-03-06:
**Describe the bug**
When I open preferences window in the IDE, the already selected theme is `Light (Arduino)`, although the active theme is `Dark (Theia)`. When I click OK, suddenly the theme `Light (Arduino)` becomes active.

**To Reproduce**
Steps to reproduce the behavior:
1. Open IDE
2. Change theme to `Dark (Theia)` using command pallete (<kbd>F1</kbd>) if it is not active
3. Open preferences
4. Field `Theme`'s value is `Light (Arduino)`

**Expected behavior**
The `Theme` field in preferences should be the same as active theme

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 1809
 -  Version: 2.0.0-beta.3
  Date: 2021-02-26T10:48:13.025Z
  CLI Version: 0.16.1 alpha [76f55490]

ubidefeo, 2021-03-19:
@Mammad900 I'm unable to reproduce with Beta 4, could you please test?

Mammad900, 2021-03-19:
I can still reproduce it on 2.0.0-beta.4

[image]

What I did:
1. Set theme to light on preferences window
2. Set theme to dark using command palette
3. Open preferences window

ubidefeo, 2021-03-20:
@Mammad900 
My bad!
I actually failed to try using the command line, which for me makes a lot of sense, being a shortcut person.
Got it now, let me add it to the next development sprint :)

thanks again for explaining twice, sometimes that's what I need 😀

Mammad900, 2021-04-04:
Fixed by #286

closed 2021-04-04 by Mammad900