arduino/arduino-ide #182: itoa not declare in scope

mrbbp, 2021-03-06:
**Describe the bug**
When i attempt to compile a skech with an itoa instruction with STM32 (not the official STM32 one but Dan Down version)
https://dan.drown.org/stm32duino/package_STM32duino_index.json
i've got an error on this 
```
/xxxx/test_itoa/test_itoa.ino: In function 'void loop()':
/xxxx/test_itoa/test_itoa.ino:25:19: error: 'itoa' was not declared in this scope
   itoa(num,snum,10);
                   ^
Compilation error: Error: 2 UNKNOWN: exit status 1
```
my script
```
int compteur=0;
char nom[] = ".raw";
int num = 0;
char snum[3];
int numOld = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  if (random(0,4)==0) {
    while(numOld == num) {
      num = random(0,21);
    }
  } else {
    num = compteur;
    compteur++;
    compteur = compteur%22;
  }
  numOld = num;
  itoa(num,snum,10);
  char chaine[8] = {};
  strncat(chaine,snum,3);
  strncat(chaine,nom,7);
  Serial.println(chaine);
  delay(1000);
  
}
```
**To Reproduce**
Steps to reproduce the behavior:
1. copy the script
2. add any STM32 Board
3. Compile

**Expected behavior**
On IDE 1.8.10 it compile perfectly with the same package_index

**Desktop (please complete the following information):**
 - OS: Macos 
 - Version: 10.14.6

per1234, 2021-03-06:
Hi @mrbbp. Thanks for your report.

I'm not able to reproduce your results with Arduino IDE 1.8.10. I get the same error either way.
Please do this:
1. Start Arduino IDE 1.8.10
1. Click **File > Preferences**
1. Check the box next to "Show verbose output during: > compilation"
1. Click "OK"
1. **Sketch > Verify/Compile**
1. After the compilation finishes, copy the full contents of the black console pane at the bottom of the Arduino IDE window.
1. Paste the output in a reply here.

mrbbp, 2021-03-08:
```

/Applications/Arduino 1.8.10.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino 1.8.10.app/Contents/Java/hardware -hardware /Users/ericchoisy-bernard/Library/Arduino15/packages -hardware /Users/ericchoisy-bernard/Dropbox/Arduino/hardware -tools /Applications/Arduino 1.8.10.app/Contents/Java/tools-builder -tools /Applications/Arduino 1.8.10.app/Contents/Java/hardware/tools/avr -tools /Users/ericchoisy-bernard/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 1.8.10.app/Contents/Java/libraries -libraries /Users/ericchoisy-bernard/Dropbox/Arduino/libraries -fqbn=Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10810 -build-path /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino_build_630407 -warnings=all -build-cache /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino_cache_608068 -prefs=build.warn_data_percentage=75 -verbose /Users/ericchoisy-bernard/Dropbox/Arduino/test_IDE1.8_ITOA/test_IDE1.8_ITOA.ino
/Applications/Arduino 1.8.10.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino 1.8.10.app/Contents/Java/hardware -hardware /Users/ericchoisy-bernard/Library/Arduino15/packages -hardware /Users/ericchoisy-bernard/Dropbox/Arduino/hardware -tools /Applications/Arduino 1.8.10.app/Contents/Java/tools-builder -tools /Applications/Arduino 1.8.10.app/Contents/Java/hardware/tools/avr -tools /Users/ericchoisy-bernard/Library/Arduino15/packages -built-in-libraries /Applications/Arduino 1.8.10.app/Contents/Java/libraries -libraries /Users/ericchoisy-bernard/Dropbox/Arduino/libraries -fqbn=Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=DFUUploadMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10810 -build-path /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino_build_630407 -warnings=all -build-cache /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino_cache_608068 -prefs=build.warn_data_percentage=75 -verbose /Users/ericchoisy-bernard/Dropbox/Arduino/test_IDE1.8_ITOA/test_IDE1.8_ITOA.ino
Using board 'genericSTM32F103C' from platform in folder: /Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1
Using core 'maple' from platform in folder: /Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1
Detecting libraries used...
/Users/ericchoisy-bernard/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10810 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple/include -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple/stm32f1/include -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/stm32f1 -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/usb_lib -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/cores/maple -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/variants/generic_stm32f103c /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino_build_630407/sketch/test_IDE1.8_ITOA.ino.cpp -o /dev/null
Generating function prototypes...
/Users/ericchoisy-bernard/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10810 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple/include -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple/stm32f1/include -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/stm32f1 -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/usb_lib -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/cores/maple -I/Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1/variants/generic_stm32f103c /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino_build_630407/sketch/test_IDE1.8_ITOA. [... cut ...] 991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino_build_630407/test_IDE1.8_ITOA.ino.bin
/Users/ericchoisy-bernard/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-size -A /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino_build_630407/test_IDE1.8_ITOA.ino.elf
Sketch uses 14224 bytes (21%) of program storage space. Maximum is 65536 bytes.
Global variables use 2200 bytes (10%) of dynamic memory, leaving 18280 bytes for local variables. Maximum is 20480 bytes.

```

mrbbp, 2021-03-08:
ok, i found the difference.

in **1.8.10 IDE** it use the **Roger Clark version** which have not a package_index.json to install but is seen by IDE : 
- https://github.com/rogerclarkmelbourne/Arduino_STM32
in IDE 2.0 it use the only one seen by ide, the Dan Drown's version (who is a fork from Roger Clark's with a package_index.json file to install) and located in a different place
- https://github.com/ddrown/Arduino_STM32/blob/master/package_STM32duino_index.json

```
Using board 'genericSTM32F103C' from platform in folder: /Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27
Using core 'maple' from platform in folder: /Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27
Detecting libraries used...
/Users/ericchoisy-bernard/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10607 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/include -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/stm32f1/include -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/usb/stm32f1 -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/usb/usb_lib -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/cores/maple -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/variants/generic_stm32f103c /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino-sketch-F54C3AF7B03FDE8DFAE077EB6A39B3E1/sketch/test_itoa.ino.cpp -o /dev/null
Generating function prototypes...
/Users/ericchoisy-bernard/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10607 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/include -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/stm32f1/include -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/usb/stm32f1 -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/usb/usb_lib -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/cores/maple -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/variants/generic_stm32f103c /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino-sketch-F54C3AF7B03FDE8DFAE077EB6A39B3E1/sketch/test_itoa.ino.cpp -o /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino-sketch-F54C3AF7B03FDE8DFAE077EB6A39B3E1/preproc/ctags_target_for_gcc_minus_e.cpp
/Users/ericchoisy-bernard/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino-sketch-F54C3AF7B03FDE8DFAE077EB6A39B3E1/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Users/ericchoisy-bernard/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10607 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple -I/Users/ericchoisy-bernard/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2021.2.27/system/libmaple/include -I/Users/ericch [... cut ...] ketch-F54C3AF7B03FDE8DFAE077EB6A39B3E1/sketch/test_itoa.ino.cpp -o /var/folders/ts/991xyp_d0r3fd9yzxywrd7nm0000gn/T/arduino-sketch-F54C3AF7B03FDE8DFAE077EB6A39B3E1/sketch/test_itoa.ino.cpp.o
/Users/ericchoisy-bernard/Dropbox/Arduino/test_itoa/test_itoa.ino: In function 'void loop()':
/Users/ericchoisy-bernard/Dropbox/Arduino/test_itoa/test_itoa.ino:25:19: error: 'itoa' was not declared in this scope
   itoa(num,snum,10);
                   ^
Compilation error: Error: 2 UNKNOWN: exit status 1
```

per1234, 2021-03-08:
OK, this is very useful information. I can see that you have a copy of the STM32 boards platform installed in your sketchbook:
>```
>Using board 'genericSTM32F103C' from platform in folder: /Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1
>Using core 'maple' from platform in folder: /Users/ericchoisy-bernard/Dropbox/Arduino/hardware/Arduino_STM32-master/STM32F1
>```

Sketchbook installed boards platforms are not currently supported by Arduino IDE 2.0.0 (https://github.com/arduino/arduino-ide/issues/126). So I'm guessing you also have a Boards Manager installation of the platform which Arduino IDE 2.0.0 is using. The use of the different boards platforms of different versions is likely the cause of the different results.

Now I would like to compare the output from Arduino IDE 2.0.0. Please do this:
1. (In the Arduino IDE) click **File > Preferences**
1. Check the box next to "Show verbose output during: > compilation"
1. Click "OK"
1. Click **Sketch > Verify/Compile**
1. Wait for the compilation to fail.
1. Right click on the black "Output" pane at the bottom of the Arduino IDE 2.x window.
1. From the context menu, click "Copy All".
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here [using code fencing](https://guides.github.com/features/mastering-markdown/).

per1234, 2021-03-08:
Oops, you're ahead of me. Haha! OK, I'm glad we have found the cause already. Since we already have a bug report about the lack of support for the sketchbook-installed boards platforms I'll go ahead and close this. You're welcome to subscribe to https://github.com/arduino/arduino-ide/issues/126 to get notifications of the progress on adding support for them.

Regards, Per

closed 2021-03-08 by per1234