arduino/arduino-ide #184: Please make Examples a top level menu

mirije, 2021-03-07:
**Is your feature request related to a problem? Please describe.**
Examples menu is nested under File menu, and itself has examples under submenus. When learning, you often want to browse multiple examples, and opening multiple hierarchy levels can get very tedious. 

**Describe the solution you'd like**
Examples menu could be a top-level menu, like File, Sketch and others. 
There are not many top level menus so I don't think this would clutter the UI too much, while at the same time it would make browsing examples much quicker.

ubidefeo, 2021-03-07:
@mirije 
thank you for your suggestion.
we have aligned the UI with the legacy Java IDE, and opening files belongs to the "File" menu.
This is proven paradigm which has been so for many years.
If your suggestion gathers consensus we'll consider it

Thank you
ubi

kwdaily, 2021-04-27:
so then maybe a duplicate version on top level? Agreed this could be useful