arduino/arduino-ide #185: Sketchesd in alphabetical order but capitalised are grouped first

ubidefeo, 2021-03-08:
**Describe the bug**
When the sketchbook menu is expanded all the ones which have capitalised names come as a group in alphabetical order, followed by a group of examples with lowercase names as another group (see screenshot)

**To Reproduce**
Steps to reproduce the behavior:
1. **File > Sketchbook**
2. Scroll examples
3. Notice that uppercase named sketches are grouped first

**Expected behavior**
Sketches should be in true alphabetical order, despite their initial

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS 10.15.7
 - Version: 2.0.0-beta.3 Date: 2021-02-26T10:47:47.411Z

silvanocerza, 2021-03-10:
I think that list is generated directly by the IDE, it's not a concern of the CLI as of now.

ubidefeo, 2021-03-12:
thanks @silvanocerza 
@kittaakos can you look into this?

Sebastian-Brzuzek, 2021-03-15:
I'm new here, so I'm not sure if proposed solution is reasonable, but if looks like assing "order" property with lower case "label" value in sketches/examples solves the issue without impacting other elements. (I have tested solution on Windows only)

ubidefeo, 2021-03-19:
Closed by #222

closed 2021-03-19 by ubidefeo