arduino/arduino-ide #186: .ino Files unable to open

JustAProgrammerr, 2021-03-08:
**Describe the bug**
A clear and concise description of what the bug is.

I can't open my .ino files.  When I click "open with" it's just an alternative for the boot up for the IDE not literally opening that specific .ino file, I have to always CTRL+O from the IDE just to open my files, not when I'm at my desktop opening a .ino file.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Desktop
2. Click on .ino files
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

To open files like the earlier, non-problematic version of the IDE

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: Beta 2.0.0

**Additional context**
Add any other context about the problem here.

Still has a good amount of features, and I love the new design, much better than the old versions.

ubidefeo, 2021-03-08:
hi @JustAProgrammerr 
this issue has already been reported and is being tracked as #72 
We have yet to implement OS file association

Thank you
u.

ubidefeo, 2021-03-19:
Closed as duplicate of #72

closed 2021-03-19 by ubidefeo