arduino/arduino-ide #187: Can't find 'esp8266 by esp8266 community'

bobblkabb, 2021-03-08:
**Describe the bug**
I have added the esp8266 url in the settings but the board manager does not find the library I need 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to _File_ -> _Preferences_
2. Add _https://arduino.esp8266.com/stable/package_esp8266com_index.json_ to Additional Boards Manager URL
3. Go to _Tools_ -> _Library Manager_
4. Search for 'community' or 'esp8266' for instance

**Expected behavior**
Show lib 'esp8266 by esp8266 community'

**Desktop (please complete the following information):**
 - OS: openSUSE Tumbleweed
 - Version: 1.8.13

**Additional context**
Was no problem to add esp32 _https://dl.espressif.com/dl/package_esp32_index.json_
Board is NodeMcu v3  LoLin (CH340G driver)

ubidefeo, 2021-03-08:
@bobblkabb 
are you using IDE 1.8.13?

This is the repository for IDE 2.0

bobblkabb, 2021-03-08:
Yes, I do use 1.8.13. Shall I use the new beta version?

ubidefeo, 2021-03-08:
@bobblkabb 
it really depends.
you can give it a try for sure, just remember that being Beta you might bump into some bugs

bobblkabb, 2021-03-08:
Still, can't also find it with latest beta (v2.0). Board ist plugged in, /dev/ttyUSB0 is selected.

ubidefeo, 2021-03-08:
@bobblkabb

I think you're doing it wrong, I only realised it now that I read your report a few times.
You are searching for the board support in the library manager

go to **Tools > Board > Board manager** and search for esp8266

bobblkabb, 2021-03-08:
That's it! I'm fool!
Thanks a lot.

closed 2021-03-08 by bobblkabb

ubidefeo, 2021-03-09:
no worries, @bobblkabb 
happy you got it to work 👍🏼