arduino/arduino-ide #188: Can't install any new board using the Board Manager

dahmadjid, 2021-03-08:
**Describe the bug**
Arduino IDE 2.0 fails to install any core using Board Manager. not even the default Arduino AVR Boards.
The output is ```
downloading arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7
downloading arduino:avrdude@6.3.0-arduino17
```
but otherwise nothing happens and no download actually starts

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Board Manager
2. Click on Install on any library

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
PS: The IDE actually recognized esp8266 core and boards like nodemcu that i previously installed on the official arduino ide 1.8.13

ubidefeo, 2021-03-09:
@dahmadjid 
can you select an AVR board and compile the code?
These messages actually tell you that it's being downloaded and what you miss is a confirmation of download being successful.
We need to do some work on the messages but I believe the core is installed at the point when your pop-up disappears.
Also IDE 2.0 recognises the already installed cores so the installation message is quick because it doesn't need to install anything, just checking for it.

>Go to Board Manager
Click on Install on any library

Please rework the wording, you do not install libraries through the Boards Manager

dahmadjid, 2021-03-09:
I can not select any board as it is not displayed (eg:Arduino Uno) so compilation is impossible.

using the task manager I can confirm that the download actually does not start. also no matter how much i wait the boards did not get installed.

ubidefeo, 2021-03-09:
@dahmadjid 
can you post screenshots of your "board" submenu and what happens when you type "avr" in the board selector?
please reference my screenshots to get a clearer idea of my requests

[image]
[image]
[image]
[image]

ubidefeo, 2021-03-12:
@per1234 
any idea what the reason behind this issue could be?

per1234, 2021-03-12:
Hi @dahmadjid. Please do this:

1. Start File Explorer (AKA Windows Explorer).
1. Open the Arduino IDE's installation folder (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **Shift** key while clicking the right hand button on the mouse.
1. Click “Open PowerShell window here”. PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type `& ".\Arduino IDE"`
1. Press **Enter**.
1. Wait for the Arduino IDE to finish starting.
2. Attempt installing one of the cores via Boards Manager.
3. If you see an "Installation in progress" dialog, wait for it to disappear
1. Switch back to the PowerShell window.
1. Right click on the title bar of the PowerShell window.
1. Click **Edit > Select All**. This will select all the text in the PowerShell window.
1. Press **Ctrl + c**. This will copy the contents of the PowerShell window to the clipboard.
1. In this GitHub issue thread, click on the reply field.
1. Press **Ctrl + V**. This will paste the output copied from PowerShell.
1. Click the "Comment" button.

Hopefully that will give us some information we can use to help you with the problem.

dahmadjid, 2021-03-16:
Sorry for the late reply, before trying what you asked for i updated the IDE to the latest nightly build and it seems to have solved the problem. Boards are getting installed with success. Thank you for the help!

closed 2021-03-16 by ubidefeo