arduino/arduino-ide #191: Cannot compile any code for OpenCM 9.04 board

tmxkn1, 2021-03-09:
The IDE 2.0 fails to compile any code for OpenCM 9.04, even a "blank" code:
```
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

```

The error I get is:
```
arm-none-eabi-gcc: error: 	-c: No such file or directory
Compilation error: Error: 2 UNKNOWN: exit status 1

```

IDE version:
```
Version: 2.0.0-beta.3
Date: 2021-02-26T10:47:31.812Z (1 week ago)
CLI Version: 0.16.1 alpha [76f55490]
```

OS:
PC1: Ubuntu 16.04.7 LTS x86064
PC2: Win10 10.0.19042

ubidefeo, 2021-03-12:
@tmxkn1 
I have tried installing the core both via IDE and CLI and I get the same failure.

I think this has to do with how the core recipes are handled by the CLI, but can't explain what's happening.
Let us look into it next week and we'll get back to you.

@cmaglie @silvanocerza I'll need you to look into why retrieval of the tools fails on CLI and not in Java IDE
the package index can be found here and correctly installs (ran CLI install in verbose mode and no failure occurred)
https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCM9.04/master/arduino/opencm_release/package_opencm9.04_index.json

Java IDE finds `arm-none-eabi-gcc` in the `packages/OpenCM904/tools/opencm_gcc/5.4.0-2016q2/bin` folder but CLI does not

per1234, 2021-03-12:
It's not that it can't find `arm-none-eabi-gcc`. It's that `arm-none-eabi-gcc` can't find a file named `	-c`.

The cause is this true tab in the C compilation recipe:
https://github.com/ROBOTIS-GIT/OpenCM9.04/blob/1.5.0/arduino/opencm_arduino/opencm9.04/platform.txt#L55
[image]

ubidefeo, 2021-03-13:
@per1234 of course!
it was too late for me to make sense of the error 🤦🏼 

so it's a whitespace issue?
why wouldn't it bother the Java IDE builder?
I'm so thankful for your mad skills in tracking down these things 😬