arduino/arduino-ide #192: When installing library through the library manager, it does not show to include in the code and does not show examples

MicSG-dev, 2021-03-10:
When installing a library through the library manager, it does not appear in the Sketch -> Include Library menu to include the library in the code. There are also no examples from the library in the File -> Examples menu.

I installed the arduino IDE 2.0 by zip. I don't know if it interfered with anything.

kittaakos, 2021-03-11:
> it does not appear in the Sketch -> Include Library menu to include the library in the code

Did you select a board, @MicSG-dev? You have to select a board to populate the `Include Library` menu. Under the hood, the IDE uses the CLI's `lib list` command filtered by the FQBN of the currently selected board. If you do not select a board, the IDE shows nothing.

https://user-images.githubusercontent.com/1405703/110769359-62211400-8258-11eb-8dee-24cbd660e29a.mp4

MicSG-dev, 2021-03-11:
Got it. Thanks for the clarification.

closed 2021-03-11 by MicSG-dev