arduino/arduino-ide #193: When booting arduino IDE 2.0, image is covering the middle of the screen

MicSG-dev, 2021-03-10:
When booting arduino IDE 2.0, an image is covering the middle of the screen for a long time, which makes it difficult to do other things while loading. Also the IDE does not load until you return to the Arduino IDE screen and click on something.

[image]

motley197, 2021-03-12:
I get the same thing - but the IDE doesn't open fully, it stalls with Windows rotating dots in the centre. The menu system is active but nothing happens if I make a selection. The IDE window responds to the close button in the top right corner.

Windows 10 Pro latest version, IDE downloaded today 12/03/2021

MicSG-dev, 2021-03-12:
@ motley197 This also occurs to me. Just by clicking anywhere on the Arduino IDE, it will be fully loaded.

motley197, 2021-03-13:
Hi, Sadly not in my case, the Help menu functions correctly as does File - Quit but all other menu choices don't work and the image you mentioned in your first post remains as well as the spinning dots.

ubidefeo, 2021-03-19:
@MicSG-dev 
are you on Windows 7 or Vista?

motley197, 2021-03-19:
Windows 10 Pro (well actually it's the Education edition but I think that's functionally the same as Pro), I tried the nightly build (.zip) and the MSI just a few minutes ago, still the same problem

MicSG-dev, 2021-03-19:
I'm on Windows 7 Ultimate 64 bit

motley197, 2021-03-19:
Do you still have the same problem as well?

MicSG-dev, 2021-03-19:
Yes, the IDE image still appears in the middle of the screen, making it difficult to view other programs, and the IDE will only finish loading when I click elsewhere in the IDE 2.0 window.

ubidefeo, 2021-03-20:
Thanks @MicSG-dev 
We'll look into it and report back.
We hope to find a fix soon

motley197, 2021-03-20:
For me the IDE never finishes loading, some parts of the menu system respond some don't.

ubidefeo, 2021-03-20:
@motley197 have you run the installer exe or manually installed?
When you first launch you should get an alert asking you to allow `arduino-cli` to run: have you allowed that?
Also, very important, which OS are you on?

motley197, 2021-03-20:
Hi I'm on Windows 10 Pro, I have tried the nightly zip (just unzipped and ran the exe file) and both the auto installers. I don't recall any mention of the arduino-cli. Let me try again...

motley197, 2021-03-20:
The current version is from the MSI yesterday, when I start the IDE I get the graphic mentioned by the OP followed shortly by the IDE with the spinning dots. Some of the menu selections work to a degree some do not. The IDE stays like this until I close it. Would you like me to try a different install?

ubidefeo, 2021-03-20:
@motley197 it might be useful to run it via the command line and see if something weird pops up.

Can you try launching the command prompt and going to the directory where your `Arduino IDE.exe` is?
I have done this

```
cd C:\Users\ubi\AppData\Local\Programs\Arduino IDE\
```
and then
```
"Arduino IDE.exe"
```

copy and paste the output of the terminal so we can look at any potential issue arising

motley197, 2021-03-20:
Here is the terminal output - it then repeats the last section over and over:

C:\Users\chris\AppData\Local\Programs\arduino-ide>
Starting backend process. PID: 27284
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root WARN please install @theia/electron@1.11.0-next.c9db9754 as a runtime dependency
root INFO Configuration directory URI: 'file:///c%3A/Users/chris/.theia'
root INFO Theia app listening on http://localhost:3071.
root WARN The local plugin referenced by local-dir:/c%3A/Users/chris/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/chris/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\chris\.arduinoProIDE\plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.7" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.1-beta.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO Deploying backend plugin "cortex-debug@0.3.7" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\cortex-debug\extension\dist\extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.1-beta.1" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-arduino-tools\extension\dist\bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-cpp\extension"
root INFO Deploying backend plugin "json@1.46.1" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-json\extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-json-language-features\extension\client\dist\jsonMain"
root INFO Deploy plugins list took: 63.1 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 233.8 ms
daemon INFO Starting daemon from C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
daemon INFO Using config file: c:\Users\chris\.arduinoIDE\arduino-cli.yaml
daemon INFO arduino-cli.exe version 0.17.0
daemon INFO Failed to listen on TCP port: 50051. Unexpected error: listen tcp 127.0.0.1:50051: bind: Only one usage of each socket address (protocol/network address/port) is normally permitted.
daemon INFO Failed to start the daemon.
daemon ERROR Error: Failed to listen on TCP port. Address already in use.
    at new DaemonError (C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\node_modules\arduino-ide-extension\lib\node\arduino-daemon-impl.js:370:28)
    at Function.parse (C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\node_modules\arduino-ide-extension\lib\node\arduino-daemon-impl.js:388:24)
    at Socket.<anonymous> (C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\node_modules\arduino-ide-extension\lib\node\arduino-daemon-impl.js:306:53)
    at Socket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:181:23)
daemon INFO Restarting daemon in 5 seconds...
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon ERROR Failed to listen on TCP port: 50051. Unexpected error: listen tcp 127.0.0.1:50051: bind: Only one usage of each socket address (protocol/network address/port) is normally permitted.
daemon INFO Daemon exited with exit code: 1.
daemon INFO Restarting daemon in 4 seconds...
root INFO Git askpass helper is listening on http://127.0.0.1:3076.
root WARN Frontend CommonFrontendContribution.configure is slow, took: 377.9 ms
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/cookie
root WARN Failed to resolve module: @types/express
root WARN Failed to resolve module: @types/fs-extra
root WARN Failed to resolve module: @types/lodash.debounce
root WARN Failed to resolve module: @types/lodash.throttle
root WARN Failed to resolve module: @types/react
root WARN Failed to resolve module: @types/react-dom
root WARN Failed to resolve module: @types/react-virtualized
root WARN Failed to resolve module: @types/route-parser
root WARN Failed to resolve module: @types/safer-buffer
root WARN Failed to resolve module: @types/ws
root WARN Failed to resolve module: @types/yargs
root WARN Failed to resolve module: @theia/monaco
root WARN Failed to resolve module: @theia/editor
root WARN Failed to resolve module: @types/rimraf
root WARN Failed to resolve module: @types/tar-fs
root WARN Failed to resolve module: @types/uuid
[... cut ...]
daemon INFO Starting daemon from C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
daemon INFO Using config file: c:\Users\chris\.arduinoIDE\arduino-cli.yaml
daemon INFO arduino-cli.exe version 0.17.0
daemon INFO Failed to listen on TCP port: 50051. Unexpected error: listen tcp 127.0.0.1:50051: bind: Only one usage of each socket address (protocol/network address/port) is normally permitted.
daemon INFO Failed to start the daemon.

ubidefeo, 2021-03-20:
> root WARN please install @***@***.*** as a runtime dependency
> root INFO Resolved "cortex-debug" to a VS Code extension ***@***.***" with engines:
> root INFO Resolved "vscode-arduino-tools" to a VS Code extension ***@***.***" with engines:
> root INFO Resolved "vscode-builtin-cpp" to a VS Code extension ***@***.***" with engines:
> root INFO Resolved "vscode-builtin-json" to a VS Code extension ***@***.***" with engines:
> root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension ***@***.***" with engines:
> root INFO Deploying backend plugin ***@***.***" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\cortex-debug\extension\dist\extension"
> root INFO Deploying backend plugin ***@***.***" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-arduino-tools\extension\dist\bundle"
> root INFO Deploying backend plugin ***@***.***" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-cpp\extension"
> root INFO Deploying backend plugin ***@***.***" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-json\extension"
> root INFO Deploying backend plugin ***@***.***" from "C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\plugins\vscode-builtin-json-language-features\extension\client\dist\jsonMain"
> at Socket. (C:\Users\chris\AppData\Local\Programs\arduino-ide\resources\app\node_modules\arduino-ide-extension\lib\node\arduino-daemon-impl.js:306:53)

As I suspected, you have another app using port 50051.

Find the arduinoIDE folder inside your AppData folder.
In there you have a file named `arduino-cli.yaml`.
Edit it and set the daemon port to something else, for instance 50059.
Quit and try again

ubi

motley197, 2021-03-20:
Hi Ubi,

Thanks very much that worked - the IDE fires up fine now :-) Interestingly the arduino-cli.yaml file had been put in the c:\users\chris\.arduinoIDE folder.

Now on to have play with the new interface :-)

Thanks again,

Chris

kittaakos, 2021-04-12:
> Thanks very much that worked - the IDE fires up fine now :-) Interestingly the arduino-cli.yaml file had been put in the c:\users\chris.arduinoIDE folder.

Correct. That is the expected location of the CLI config.

Related: https://github.com/arduino/arduino-ide/issues/201

Wolfy7, 2021-08-10:
Hello all.

I also have this issue, maybe related to #385 
see my comment there:
https://github.com/arduino/arduino-ide/issues/385#issuecomment-895861208

0ljik, 2022-02-25:
I am also having the same issue with 2.0.0-rc3 on windows 11

stodge, 2022-05-26:
This still happens on rc6. When the splash screen is displayed, it hides everything else so I can't use another application.

MicSG-dev, 2022-05-26:
> Contribuinte

I'm still affected by this issue in the latest version of Arduino IDE 2.0;

stodge, 2022-06-10:
rc7 is exactly the same for me, no improvement at all.

commit 0e4a6a6c (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #193
Closes #324
Closes #327
Closes #717
Closes #851
commit f7a13539 (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #193
Closes #324
Closes #327
Closes #717
Closes #851
commit 0adde9a9 (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #193
Closes #324
Closes #327
Closes #717
Closes #851
commit db9037dc (2022-08-04): Replaced the splash screen with a preload.

Added a bare minimum example.

Closes #193
Closes #324
Closes #327
Closes #717
Closes #851
closed 2022-08-09 by kittaakos