arduino/arduino-ide #194: Auto Format function does not work

MicSG-dev, 2021-03-10:
The Auto Format function, triggered by the shortcut CTRL + T or the Tools -> Auto Format menu, is not working. When triggering Auto Format, nothing happens to the code when it is badly formatted.

kittaakos, 2021-03-11:
This issue a duplicate of https://github.com/arduino/arduino-ide/issues/83. Closing. @MicSG-dev, let me know if you disagree and I am happy to discuss it.

closed 2021-03-11 by kittaakos

MicSG-dev, 2021-03-11:
In [# 83] (https://github.com/arduino/arduino-ide/issues/83), it says that "Automatic formatting seems to work only on INO files; nothing happens when applied to h or cpp files". But my problem is that automatic formatting for INO files does not work.

I installed the arduino IDE 2.0 by zip.

kittaakos, 2021-03-11:
> In #83, it says that "Automatic formatting seems to work only on INO files; nothing happens when applied to h or cpp files". But my problem is that automatic formatting for INO files does not work.
> 
> I installed the arduino IDE 2.0 by zip.

Thanks for the feedback, you're right. I have linked the incorrect issue. Here is the correct one: https://github.com/arduino/arduino-ide/issues/78

MicSG-dev, 2021-03-11:
ok

kittaakos, 2021-03-11:
The fix will be part of the next beta release for sure. The [changes](https://github.com/arduino/arduino-language-server/pull/69) are all ready for the merge. Please bear with us. Thanks!

MicSG-dev, 2021-03-11:
Don't worry, take as much time as you need. I'm happy to help.