arduino/arduino-ide #199: Shift + Upload does not use programmer

toogooda, 2021-03-10:
**Describe the bug**
Old Arduino IDE used the programmer for uploading when you held down shift before the upload button. IDE 2.0 does not and only works for hot keys CTRL + SHIFT + U.

**To Reproduce**
Steps to reproduce the behavior:
1. Select a programmer
2. Hold down Shift and click upload

**Expected behavior**
Same as previous IDE as there are so many tutorials that all tell you to use SHIFT+Upload Button.
Holding shift and hovering over upload should change the hit from "Upload" to "Upload with Programmer" 
Holding shift and clicking Upload should do the same at CTRL+SHIFT+U

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 10

kittaakos, 2021-03-11:
Thank you for reporting the issue, @toogooda. The request is tracked [here](https://github.com/arduino/arduino-ide/issues/103#issue-818838087):

> * It is confusing because there is no symbol in the Pro IDE for "Upload Using Programmer"
>   In the standard IDE there is only one button for both, but also two commands.

I am closing this as a duplicate.

per1234, 2023-02-21:
After the dust settled, https://github.com/arduino/arduino-ide/issues/103 was determined to be on the topic of the missing support for this feature:

https://arduino.github.io/arduino-cli/dev/platform-specification/#upload-using-programmer-by-default

I didn't find any other issues tracking the missing support for the <kbd>**Shift**</kbd>+Upload button combination from Arduino IDE 1.x so I'll reopen this.

If implementing the feature is determined to be technically infeasible, or not needed from a UX standpoint then that decision can be documented here before closing again.