arduino/arduino-ide #200: Board and port selected always shows with warning icon

toogooda, 2021-03-10:
**Describe the bug**
Every board I select always shows with a warning icon next to it, even though upload works and so does output.

**To Reproduce**
Steps to reproduce the behavior:
1. Select any valid board and com port
4. See warning icon

**Expected behavior**
No idea as there is nothing to tell you what it is for, it shows a red 'x' when nothing is connected so I assume a green icon or nothing when a arduino is connected.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS:  Windows
 - Version: 10

**Additional context**
I wasted time trying to work out what I had wrong because the warning icon indicated a potential issue.
Tried on multiple Arduino compatible devices.

kittaakos, 2021-03-11:
Thanks for reporting, @toogooda. It's the expected behavior. You can see [here](https://github.com/arduino/arduino-pro-ide/issues/198#issuecomment-600059664) how it works. In a nutshell: if the Arduino CLI cannot recognize your board, no `fqbn` is available via the `board list` command, you will see a warning.

This is the CLI's output when I have attached a MKR1000 and ESP8266 board. You can see the details for the ESP8266 is not available.

```
./arduino-cli board list --format json
[
  {
    "address": "/dev/cu.URT2",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.URT1",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.Bluetooth-Incoming-Port",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.usbserial-0001",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)"
  },
  {
    "address": "/dev/cu.usbmodem14401",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)",
    "boards": [
      {
        "name": "Arduino MKR1000",
        "FQBN": "arduino:samd:mkr1000",
        "VID": "0x2341",
        "PID": "0x804E"
      }
    ]
  }
]
```

I am closing as invalid. Let me know if you disagree, @toogooda. I am happy to discuss it further. Thanks!

toogooda, 2021-03-11:
Hi kittaakos, 
Thanks for your explination. by far the majority of clones will use the CH340G/C which will not return a FQBN as the chip uses a default I believe. This will just be a nag or confusing(what have I done wrong?) to everyone who uses them. Is this feature more hinderance than help?
It is helpful displaying the selected board and port, and auto populating when you can but this permanent warning seems like a step to far.
I could do a hundred successful uploads and it will still be warning me. 
Can I toggle it off?

ubidefeo, 2021-03-12:
@toogooda you are in the right.
A warning sign should be an indicator that something is wrong.

We'll look into it 👍🏼

toogooda, 2021-03-12:
> @toogooda you are in the right.
> A warning sign should be an indicator that something is wrong.
> 
> We'll look into it 👍🏼

Thank you

ckuethe, 2021-03-17:
Can there be a way to set my preferred serial port patterns? I only ever have `ttyUSB*` (rarely `ttyACM*`) and right now `ttyUSB0` is waaaaaaaaayyyyyy down at the bottom, below a bunch of `ttyS*` devices.

ubidefeo, 2021-03-20:
@ckuethe 
>Can there be a way to set my preferred serial port patterns? I only ever have ttyUSB* (rarely ttyACM*) and right now ttyUSB0 is waaaaaaaaayyyyyy down at the bottom, below a bunch of ttyS* devices.

Unfortunately cleaning up your serial ports files is up to you.
we run a serial discovery which lists every serial device on the system so if you have 50 serial ports that's what you'll get

steveboak, 2021-07-17:
I too wasted time investigating this warning before I found this post.
I would like to propose a solution: An explanatory message pop up when the mouse is hovered over the warning icon.

toogooda, 2021-07-21:
> I too wasted time investigating this warning before I found this post.
> I would like to propose a solution: An explanatory message pop up when the mouse is hovered over the warning icon.

I disagree,a warning suggests a problem and they're is not, adds there is nothing that cab bees done to fix it it is just a pregnant nag! Please remove it.

moth4017, 2021-09-08:
I agree a warning triangle is to informof an issue.

ubidefeo, 2022-04-05:
Just an update on this issue:
we're reworking the iconography for this area of the UI and this is being addressed in the process.
This issue will remain open and be connected to our internal development backlog in order to be closed by an upcoming PR

Thank you for your patience ✌🏼

per1234, 2022-07-19:
Closing as fixed by https://github.com/arduino/arduino-ide/pull/1193

closed 2022-07-19 by per1234