arduino/arduino-ide #201: Beta IDE hangs on startup due to CLI deamon continuously restarting due to port 50051 unavailble

jens-a-e, 2021-03-11:
**Describe the bug**
The Arduino IDE 2 beta GUI hangs on a rotating spinner and splashscreen when another service is listening on port 50051 (arduino cli port).

**To Reproduce**
Steps to reproduce the behavior:
1. Start a service listeing on port 50051 (IPv4 and IPv6)
2. Double click the Arduino IDE

**Expected behavior**
The Arduino IDE to start past the spinner and give a warning that port 50051 is unavaible. Possibly indicating how to change the port settings.

**Desktop (please complete the following information):**
 - OS: macOS
 - Version: 2.0.0-beta.3

**Additional context**

To mitigate the problem I started the Arduino IDE via the command line. That gave me the error of the unavailable port. I resolved the service that was listening on port 50051—Canonical's Multipass in my case. However, I was wondering if the port could be randomly picked and written/updated to the `.arduino/arduino-cli.yml` file?

kittaakos, 2021-03-11:
> However, I was wondering if the port could be randomly picked and written/updated

Thank you, @jens-a-e 👍 This is a good idea, I do not know how this could work in practice but it worth some discussion. However, this should go to the CLI repository: https://github.com/arduino/arduino-cli/issues/new

> to the `.arduino/arduino-cli.yml` file?

For the record: the new IDE keeps its config file under `~/.arduinoIDE/arduino-cli.yaml`.

> Possibly indicating how to change the port settings.

If somebody else has the same issue and is looking for a workaround: https://github.com/arduino/arduino-ide/issues/158

----

This is the request here 👇, the rest is out of the context of the new IDE.

> **Expected behavior**
> The Arduino IDE to start past the spinner and give a warning that port 50051 is unavaible. Possibly indicating how to change the port settings.

jens-a-e, 2021-03-15:
Thank you @kittaakos! I will investigate the random port and create an issue at the cli repo.

It would still be great if the IDE could provide some visual feedback if the startup does not go well.
Although changing the port in the config file of a hidden directory might work for beta users, it would help to have some GUI to easily share the error.

thughes, 2021-03-20:
I also hit this issue immediately after installing the IDE. Unfortunately the behavior made it look like the IDE had just frozen and it wasn't obvious what to do. When running from the commandline I saw the logs complaining about port `50051` being in use:

```
daemon INFO Using config file: /Users/tom/.arduinoIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.17.0
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon ERROR Failed to listen on TCP port: 50051. Address already in use.
daemon INFO Failed to start the daemon.
daemon ERROR Error: Failed to listen on TCP port. Address already in use.
    at new DaemonError (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:370:28)
    at Function.parse (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:388:24)
    at Socket.<anonymous> (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/arduino-daemon-impl.js:281:57)
    at Socket.emit (events.js:223:5)
    at addChunk (_stream_readable.js:309:12)
    at readableAddChunk (_stream_readable.js:290:11)
    at Socket.Readable.push (_stream_readable.js:224:10)
    at Pipe.onStreamRead (internal/stream_base_commons.js:181:23)
daemon INFO Restarting daemon in 5 seconds...
daemon INFO Failed to listen on TCP port: 50051. Address already in use.
daemon INFO Daemon exited with exit code: 5.
daemon INFO Restarting daemon in 4 seconds...
daemon INFO Restarting daemon in 3 seconds...
daemon INFO Restarting daemon in 2 seconds...
daemon INFO Restarting daemon in 1 seconds...
daemon INFO Restarting daemon now...
```

I also have multipass installed. I ended up changing the port in `~/.arduinoIDE/arduino-cli.yaml` to `50052`, which allows both to run. Given that multipass is already using `50051`, perhaps the default for the IDE should be changed. Additionally, an error message in the IDE would be helpful so that you don't have to rely on the logs.

ubidefeo, 2021-03-20:
@thughes 
we'll look into finding a more flexible solution to this.
Certainly adding a meaningful error will help.
We're in the process of making errors more meaningful in general

thank you for chiming in
u.

kittaakos, 2021-04-12:
> I will investigate the random port and create an issue at the cli repo.

https://github.com/arduino/arduino-cli/issues/1260

jens-a-e, 2021-04-13:
Great! Thank you, much appreciated.

closed 2021-12-09 by fstasi