arduino/arduino-ide #204: vim mode

jerabaul29, 2021-03-11:
Is it possible to get a vim mode for the code edition?

ubidefeo, 2021-03-12:
@jerabaul29 I guess you mean shortcuts, right?

@kittaakos might have the answer you're looking for, although we don't officially support extended preferences in case a user bumps into a weird behaviour we don't publicly expose/document :)

jerabaul29, 2021-03-12:
Mmh, not really, was thinking about the vim modal typing flow. Like edit, visual, default modes (and all the shortcuts there, true). VSC has a vim mode available through a plugin for example.

Looks like Theia IDE may have a compatible vsx plugin that provides that. Any way this could be installed as an opt in, and / or is there a way to directly manage plugins as a user?

ubidefeo, 2021-03-12:
@jerabaul29 
what I replied in #207 also applies here.

We have to make sure this IDE works for the widest user base.
Feel free to mess around with `.vsx` plugins and try to make it work :)

This issue will be closed as **out of scope**

closed 2021-03-12 by ubidefeo