arduino/arduino-ide #205: pane for displaying errors provided by the linter / static analyzer

jerabaul29, 2021-03-11:
I am having the following "hello world" code that displays some warnings (red squiggles).

[image]

First, this is a bit surprising, as the code actually compiles fine and the error message (see later) is not really relevant. I wonder if the IDE misses some paths when looking for valid sources (I am using a custom core, from Sparkfun, for the Artemis boards).

My main question is: is there / can we get a separate pane or similar for listing all such errors? As of now it seems that the way for me to get more information is to hoover with the mouse over the squiggles (see screenshot). I wonder if for larger projects it would be easier to have a dedicated "linter and static analyzer warnings" pane, a la platformio IDE.

[image]

ubidefeo, 2021-03-12:
@jerabaul29 
thanks for reporting this.
During development we used to have a stream from the Language Server available in output, but it proved to be confusing to testing users and mostly cluttering the UI.
You can enable logging from the language server from the preferences if you don't mind going through MegaBytes of messages in a `.log` file.
It is actually how we debug the LS.

[image]

jerabaul29, 2021-03-12:
Ok, thanks, going through MBs sounds a bit inconvenient. I will wait for a more mature 'error pane' then.

closed 2021-03-12 by ubidefeo