arduino/arduino-ide #206: Autocomplete not working as described

UKHeliBob, 2021-03-11:
Following the instructions from https://www.arduino.cc/en/Tutorial/getting-started-with-ide-v2/ide-v2-autocomplete-feature/ I cannot make Autocomplete work as described

With a new sketch loaded and Arduino Nano selected as the board type, typing pin as suggested brings up an autocomplete suggestions list with pinMode at the top, and clicking on that or pressing return selects it.  uint8_t pin is automatically selected as described, but typing but typing LED does not bring up the suggestion of LED_BUILTIN as described.  Typing the name in full and hovering over it does show the fact that it is a macro with a value of 13 and right clicking and selecting  Go to Definition opens pins_arduino.h showing how the macro is defined so it is obviously known to the editor

See https://forum.arduino.cc/index.php?topic=731873

ubidefeo, 2021-03-19:
@UKHeliBob 
please follow the issue report template in order to keep all information centralised.
This helps us track issues much better without a back-and-forth between GH and Forum

thanks

UKHeliBob, 2021-03-19:
**To Reproduce**
Following the instructions from https://www.arduino.cc/en/Tutorial/getting-started-with-ide-v2/ide-v2-autocomplete-feature/ I cannot make Autocomplete work as described

With a new sketch loaded and Arduino Nano selected as the board type, typing pin  brings up an autocomplete suggestions list with pinMode at the top, and clicking on that or pressing return selects it.  uint8_t pin is automatically selected as described,

**Expected behaviour**
Typing LED at that point should bring up the suggestion of LED_BUILTIN but it does not

**Desktop Environment**
Windows 10
IDE version 2.0.0-Beta 3

**Additional context**
See https://forum.arduino.cc/index.php?topic=731873

ubidefeo, 2021-03-20:
thanks @UKHeliBob 
Sometimes macros get lost for some targets, even though similar targets with same CPU/specs find and suggest the expansion.
Language Server is still quite glitchy, but it is in constant development and refinement so I'm sure we'll nail this kind of issue as we move forward ✌🏼

ktprograms, 2021-04-09:
I'm also having this problem. The autocomplete shows up if I press ctrl+space, but not automatically. I think there might be a setting to trigger the autocomplete on more characters but I can't find it.

per1234, 2021-11-26:
> The autocomplete shows up if I press ctrl+space, but not automatically.

I think this comment is the key to understanding the issue.

I was having trouble reproducing it because I always have **File > Preferences > Editor Quick Suggestions** disabled and so I instinctively use the "Trigger Suggest" keyboard shortcut (e.g., <kbd>Ctrl</kbd>+<kbd>Space</kbd>) to trigger the autocomplete suggestions.

The tutorial was written before the "Editor Quick Suggestions" was changed to being off by default (https://github.com/arduino/arduino-ide/pull/221), so it assumes the feature is on. After enabling the feature, I get the expected quick suggestion for `pinMode`, but not for `LED_BUILTIN` argument (even while using the build from https://github.com/arduino/arduino-ide/pull/610 with the revamped language server). But this behavior is specific to the argument autocomplete process. If I type "LED" as an argument while not in an autocomplete operation, then I get the `LED_BUILTIN` quick suggestion as expected.

It also isn't specific to macros. For example, you won't get a quick suggestion for `random` in the argument while autocompleting `delay(random(1000));`

per1234, 2022-07-30:
This issue has two components:

## ["autocomplete" tutorial](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-autocomplete-feature) not aligned with actual behavior of IDE

### Assumption the user has "**Editor Quick Suggestions**" enabled

This was resolved 2021-12-07 (before the tutorial was published to the public repository) by the addition of an "**Enable/Disable Autocompletion**" section to the tutorial:

https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-autocomplete-feature#enabledisable-autocompletion

The tutorial instructions were later changed to mostly use the <kbd>**Ctrl**</kbd>+<kbd>**Space**</kbd> "**Trigger Suggest**" keyboard shortcut, making them applicable even for users with "**Editor Quick Suggestions**" disabled in their preferences: https://github.com/arduino/docs-content/pull/427

### Claim that quick suggestions are provided for auto-selected arguments

This was resolved by https://github.com/arduino/docs-content/pull/427

## IDE does not provide quick suggestions for auto-selected arguments

Since it is maybe a little more difficult to reproduce now that following the tutorial doesn't result in this behavior, I'll provide complete instructions to reproduce it:

1. Select **File > Preferences** from the Arduino IDE menus
1. If it is not already, check the box next to "**☐ Editor Quick Suggestions**".
1. Click the <kbd>**OK**</kbd> button.
1. Create a sketch with the following code:
   ```cpp
   void foo(int bar, int baz) {}
   const int pippo=42;
   void setup() {

   }
   void loop() {}
   ```
   **ⓘ** The code is different from that in the tutorial only to provide a self contained demo that doesn't rely on code from the core of the selected board. The `pinMode` code from the tutorial will also produce the issue though.
1. Select any board with language server support from the **Tools > Board** menu (e.g., "Arduino Uno")
1. In the `setup` function definition block, type the following text:
   ```cpp
   fo
   ```
1. Wait for the **Suggest** menu to open.
1. Select **foo(int bar, int baz)** from the **Suggest** menu.
   The following code will be added to the sketch:
   ```cpp
   foo(int bar, int baz)
   ```
   The `int bar` argument will be automatically selected.
1. Type `pipp`
1. Wait for the **Suggest** menu to open.
   🐛 The **Suggest** menu never opens.
   [image]
1. Press <kbd>**Ctrl**</kbd>+<kbd>**Space**</kbd>
   🙂 The **Suggest** menu opens as expected.
1. Move the cursor somewhere outside the `foo` function call in the Arduino IDE editor.
   The gray highlights on the function call arguments disappear.
1. Type `pipp` again in the first argument of the `foo` function call.
1. Wait for the **Suggest** menu to open.
   🙂 The **Suggest** menu opens as expected.
   [image]

Since I don't ever use code completion, I'm no expert on how the **Suggest** feature is expected to work, but this behavior seems quite inconsistent and annoying to me. However, I see that it also occurs when I do the same procedure with a pure C++ program in VS Code with the [**clangd** extension](https://github.com/clangd/vscode-clangd) installed. So maybe it is the expected and correct behavior? In that case, we could close this as resolved by the tutorial changes I mentioned above.

I didn't find a setting that allows this behavior to be adjusted.

ketwaroo, 2023-07-13:
I have a similar issue, may be related to this? https://forum.arduino.cc/t/solved-auto-completion-not-working-on-arduino-2-1/1120524

The solution described there was to move all the sketched to C: drive. That did not work for me. I'm not sure if the Arduino IDE itself needs to be installed on C: drive on windows.

For me function completion is missing. if I type `mil` and press `Ctrl+Space`, I expect it to autocomplete to `millis()` and with all the param hints/overloads/etc if any. 

The weird part it was working a few days ago. The only change is that I had arduino IDE installed at the same time. I did some cleanup and removed all Arduino related software and reinstalled only IDE 2.1.1 and libraries for it.

ketwaroo, 2023-07-13:
So a solution presented itself not that I took 2 minutes to think about it.

I downloaded the latest release of `arduino-cli:0.33.1` and replaced the `arduino-cli.exe` under `resources\app\node_modules\arduino-ide-extension\build`.  Quick suggestions now work  as expected.

Seems that CLI 0.33.1 will be bundled with IDE 2.1.2 so if sketches are stored on a different drive than C:, that problem will resolved itself. Not sure if the exact same problem as @UKHeliBob though.