arduino/arduino-ide #207: Supporting open vsx plugins and extensions

jerabaul29, 2021-03-12:
I think that, being based on Theia, this IDE should be compatible more or less out of the box with the open vsx ecosystem, right?

https://www.gitpod.io/blog/open-vsx/

Can we get a dedicated menu / pane to the Debug / Boards Manager / Library Manager ones, but dedicated to user installed plugins and extensions, to be able to install such extensions and plugins manually?

ubidefeo, 2021-03-12:
@jerabaul29 
I believe advanced users who want to customise the IDE to their likings should do so, but making this a public function would mean we had to support it and all the quirks that might come from installing any random plugin.

As a small team we have to pick and choose what we can afford to maintain and need to reduce the amount of potential issues flooding this space.

I'm sure if you look at advanced preferences you can figure out how to enable such support, but we cannot help fix any issue caused by tempering with the IDE.

Hope you'll understand :)

jerabaul29, 2021-03-12:
Do you really not want to make it possible for the users to control extensions? That would be quite sad, this is the whole point of having an ecosystem IMO :( . I agree that you cannot support what your users do with the IDE when adding extensions, but I think it is a bit radical to 'discourage' users from customizing by not making it possible.

ubidefeo, 2021-03-12:
@jerabaul29 not discouraging
that is far from our intentions.
hacking is great :)

we will eventually end up adding an **advanced > obscure > black magic > pray before using ** preference hierarchy somewhere 😆 

making it part of the main UI will be confusing to beginners who will end up enabling/installing things that they don't know about and running to the forums to ask for help.

Remember that while users like us have no problem using VS Code, Sublime Text, XCode or Eclipse, a beginner can be scared away pretty easy, and the success of Arduino comes from simplified interfaces but powerful tools behind the curtains 😉

closed 2023-08-03 by per1234

FlightControl-User, 2023-10-26:
Which option do i select to develop VS code extensions for Arduino IDE 2?

[image]

Web extension?

per1234, 2023-10-26:
Hi @FlightControl-User. Thanks for your interest in Arduino IDE extensions! This issue tracker is only to be used to report bugs or feature requests specific to the project, and to discussions tightly focused on the implementation of the fixes and features in the codebase. Assistance with making 3rd party extensions is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/c/software/ide-2-x/93