arduino/arduino-ide #210: IDE 2.0 not work Portable mode (portable folder)

sergiopoverony, 2021-03-12:
**Describe the bug**
When creating the folder name as portable, portable mode was enabled in the 1st version. This function does not work in this version.

**To Reproduce**
Steps to reproduce the behavior:
1. make folder name as portable 

**Expected behavior**
Sketchbook and libraries are added to the portable folder

**Screenshots**
no need

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
Need portable mode ;)

per1234, 2021-04-19:
Hi @sergiopoverony. Thanks for taking the time to submit an issue.

I see we already have a prior issue report about this at https://github.com/arduino/arduino-ide/issues/122.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-04-19 by per1234

njordan77, 2023-09-02:
Any update. not sure if the group of developers do understand the need. but for long time now people kindly asked to make it happen like it was possible in 1.x IDE. So please read all the frustrated comments on the internet, people have a really high demand for this. Thanks a lot