arduino/arduino-ide #214: No print option available

UKHeliBob, 2021-03-13:
Whilst it might seem archaic to want to print code, it comes into its own for examination of the code away from a PC, and can also be useful in providing documentation of a project.  This is particularly the case where it is "printed" to a PDF file

Karl255, 2021-03-13:
@UKHeliBob 
There's a workaround for this. First, select the part of the code you want, then open the command palette with F1 or Ctrl + Shift + P, type "Copy With Syntax Highlighting" and press enter. Now you can paste the code into an editor like Word or Google Docs and print from there. Don't forget to set the font to a monospace font afterwards (such as Consolas or Courier New).

Lots of modern code editors support this feature in their own way.

UKHeliBob, 2021-03-14:
@Karl255

Thank you for the reply but I am well aware of how to do a copy and paste by one or more means or another, Ctrl + a, Ctrl + c being the quickest and requiring no use of the mouse

The suggested workaround adds unnecessary complexity to what should be a simple process of right click, P, then OK in the print dialogue.  The workaround requires the use of both the keyboard and mouse, multiple steps in the editor and an external program

I note too that the standard Windows shortcut of Ctrl + p to open the print dialogue has been used for some kind of search and that the right click menu has no Select All entry.  Thank goodness that Ctrl + a still does Select All

"Lots of modern code editors support this feature in their own way."
In my opinion this is not a feature, rather it is an excuse for a missing feature

I realise that the functionality of the IDE editor is not directly the responsibility of the Arduino team but it really sucks at doing some simple tasks

jjohnson873, 2022-10-09:
Arduino 2.0.0....no Print function?!   This is needed for those of us who may not have a laptop available at every waking moment.

Tom55k, 2022-10-11:
I'm sorry and I don't want to be offensive but the lack of a Print function is one of the dumbest omissions that could have been made. I will not be using 2.0 until this severe deficiency is corrected. I was really starting to like it until this showed up! I will revert to 1.8 - til then...