arduino/arduino-ide #216: Unable to shrink IDE window.

Prof256, 2021-03-14:
OS: Ubuntu 20.04
Display: 1366x768
IDE: 2.0.0-beta.3

Opening state of IDE is taller than my screen and will not shrink using normal methods. Because the IDE is too tall I can not see the bottom of the program.

Different topic but a maximize button would be nice.

BelKed, 2021-03-14:
Same here.
**OS:** MacOS 10.13.6 (High Sierra)
**Display:** 1440 x 900
**IDE:** 2.0.0-beta.3

ubidefeo, 2021-03-14:
@kittaakos can we read the OS's max screen height and adjust it on launch?

per1234, 2021-03-14:
See https://github.com/arduino/arduino-ide/issues/96

abnair1982, 2021-03-17:
Due to this, the button to select the board gets hidden. And I have to drag the ide to the right. I have ios style dashboard on my ubuntu system. Screenshot below:

Before moving the ide:

[image]

After moving the ide:

[image]

Attigliuzzo, 2021-03-18:
I reported the same issue [https://github.com/arduino/arduino-pro-ide/issues/395](url) with Arduino ide pro. It isn't ever fixed, i don't test the beta4 yet.

abnair1982, 2021-03-18:
> I reported the same issue
> https://github.com/arduino/arduino-pro-ide/issues/395 <http://url> with
> Arduino ide pro. It isn't ever fixed, i don't test the beta4 yet.

It has been now merged with an existing issue.

https://github.com/arduino/arduino-ide/issues/96#issuecomment-801883221

commit 128bffa0 (2021-03-29): Removed the min window height and width.

Made the boards config dialog slightly smaller.

Closes #216.
commit 4e7f8291 (2021-03-31): Removed the min window height and width.

Made the boards config dialog slightly smaller.

Closes #216.