arduino/arduino-ide #220: Official STM32 showing many undeclared identifiers for blink sketch but compiles ok

toogooda, 2021-03-15:
**Describe the bug**
LED_BUILTIN,  HIGH, delay, OUTPUT all underlined in red and showing "Use of undeclared identifier". Strangely digitalWrite and pinMode are recognised OK.
If you ignore and compile it completes without exception.

**To Reproduce**
Steps to reproduce the behavior:
1. Add additional boards URL "https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json"
2. Select Blink example
3. Select any STM board I used STM32F103RC

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows 10

**Additional context**
There are other issue with STM boards that I will log separately.

toogooda, 2021-03-18:
It is also inconsistent in that the same variable can be used twice, once recognised and once unidentified. see image and delay examples:
[image]

ubidefeo, 2021-03-19:
@toogooda 
sometimes the definition of macros from 3rd party cores doesn't play well with the Language Server.
You can see that `digitalWrite` and `pinMode` behave correctly.

The LS is still in an early stage of refinement, and it's being actively worked on.
Bare with us

For now I'm assigning this a lower priority
thanks

toogooda, 2021-03-21:
> @toogooda
> sometimes the definition of macros from 3rd party cores doesn't play well with the Language Server.
> You can see that `digitalWrite` and `pinMode` behave correctly.
>
> The LS is still in an [...]

No problem, happy to do what testing needs doing for STM32.

ag88, 2021-03-27:
i'd like to provide a little more info on this topic, i turned on language server logging
in /tmp/arduino-language-server991402071 i found
**build.options.json**
```
{
  "additionalFiles": "",
  "builtInLibrariesFolders": "",
  "builtInToolsFolders": "",
  "compiler.optimization_flags": "",
  "customBuildProperties": "build.warn_data_percentage=75",
  "fqbn": "STM32:stm32:GenF4",
  "hardwareFolders": "/home/user/.arduino15/packages",
  "otherLibrariesFolders": "/home/user/Arduino/libraries",
  "runtime.ide.version": "10607",
  "sketchLocation": "/home/user/Arduino/sketch_mar28a/sketch_mar28a.ino"
}
```
**includes.cache**
```
[
  {
    "Sourcefile": null,
    "Include": "",
    "Includepath": "/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino"
  },
  {
    "Sourcefile": null,
    "Include": "",
    "Includepath": "/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/variants/BLACK_F407XX"
  },
  {
    "Sourcefile": "/tmp/arduino-language-server991402071/sketch/sketch_mar28a.ino.cpp",
    "Include": "",
    "Includepath": null
  }
]
```
^^^ the path of the core is correctly inferred, but that there is sub-directories with more include and c files
the include path for the variant is incorrect and the source files given as null, and includes given as ""
i'd think many paths are actually missing

**preproc/** folder empty
**sketch/** folder has the correct sketch

ag88, 2021-03-27:
couldn't really figure out how 
[vscode-arduino-tools](https://github.com/arduino/vscode-arduino-tools)
[arduino-language-server](https://github.com/arduino/arduino-language-server)
works in IDE 2.0, it seem to be calling clangd, but it is difficult to figure out how the relevant directories and files are passed

ubidefeo, 2021-03-28:
@ag88 
our language server does use Clangd and is currently actively being developed for Arduino IDE 2.0
Refinements to how it behaves inside VSCode might come in the future, but for now we do not provide official support for VS Code

ag88, 2021-03-28:
one more observation as i'm trying to figure out the root cause
**compile_commands.json**
```
[]
```
pretty much an empty file
the specs for that seem to be here
https://clang.llvm.org/docs/JSONCompilationDatabase.html
perhaps i'd try to define a compile_commands.json manually and try the autocomplete again?

apparently clangd does some form of 'background indexing'
accordingly compile_commands.json is needed 
https://clangd.llvm.org/installation.html

any hints on how to go about with this?

ag88, 2021-03-28:
i made a minor progress here. it turns out arduino-cli now generates compile_commands.json in the build folder
https://github.com/arduino/arduino-cli/issues/849
e.g. /tmp/arduino-sketch-33482DDA232A7A14.../compile_commands.json
```
[
 {
  "directory": "/opt1/arduino/arduino-ide_2.0.0-beta.4-snapshot.51195eb_Linux_64bit",
  "arguments": [
   "/opt1/arduino/dotarduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-g++",
   "-mcpu=cortex-m4",
   "-mfpu=fpv4-sp-d16",
   "-mfloat-abi=hard",
   "-mthumb",
   "@/tmp/arduino-sketch-33482DDA232A7A142FE6DC7A771F0016/sketch/build_opt.h",
user@snoopy1:/tmp/arduino-sketch-33482DDA232A7A142FE6DC7A771F0016> head -30 compile_commands.json 
[
 {
  "directory": "/opt1/arduino/arduino-ide_2.0.0-beta.4-snapshot.51195eb_Linux_64bit",
  "arguments": [
   "/opt1/arduino/dotarduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-g++",
   "-mcpu=cortex-m4",
   "-mfpu=fpv4-sp-d16",
   "-mfloat-abi=hard",
   "-mthumb",
   "@/tmp/arduino-sketch-33482DDA232A7A142FE6DC7A771F0016/sketch/build_opt.h",
   "-c",
   "-Os",
   "-w",
   "-std=gnu++14",
   "-ffunction-sections",
   "-fdata-sections",
   "-nostdlib",
   "-fno-threadsafe-statics",
   "--param",
   "max-inline-insns-single=500",
   "-fno-rtti",
   "-fno-exceptions",
   "-fno-use-cxa-atexit",
   "-MMD",
   "-I/home/user/Arduino/sketch_mar28b",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/avr",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/LL",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/OpenAMP",
```
pretty large file, i tried copying 
/tmp/arduino-sketch-33482DDA232A7A14.../compile_commands.json to 
/tmp/arduino-language-server19487.../compile_commands.json
replacing the empty file.

unfortunately that didn't solve the issue, i try clicking verify/compile but the results is quite similar
many of the symbols gives a 'no definition found' if i right clicked and select goto definition.

should i exit arduino-ide and restart 
/tmp/arduino-language-server19487.. 
folder is deleted and recreated with an empty compile_commands.json

ag88, 2021-03-29:
the language server toolchain architecture apparently seem to be
arduino-ide2 -> vscode-arduino-tools (module) -> arduino-language-server -> clangd
in **inols-err.log**
```
--> initialize(file:///home/user/Arduino/sketch_mar28b)
running:  /opt1/arduino/arduino-ide_2.0.0-beta.4-snapshot.51195eb_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli compile --fqbn STM32:stm32:GenF4 --only-compilation-database --clean --source-override /tmp/638675773 --build-path /tmp/arduino-language-server071191930 --format json /home/user/Arduino/sketch_mar28b
```
hence compile_commands.json is actually generated by arduino-cli using the --only-compilation-database option

i went into /tmp/arduino-language-server071191930 and tried
```
/opt1/arduino/arduino-ide_2.0.0-beta.4-snapshot.51195eb_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli compile --fqbn STM32:stm32:GenF4 --only-compilation-database --clean --build-path . --format json /home/andrew/Arduino/sketch_mar28b
{
  "compiler_out": "",
  "compiler_err": "arm-none-eabi-g++: error: ./sketch/build_opt.h: No such file or directory\n",
  "builder_result": {
    "build_path": "."
  },
  "success": false
}
```
^^^ in running the above command i've removed the source override option and did the compilation in the /tmp/arduino-language-server634547158 directory

however if I remove --only-compilation-database option the build completes, though with a success : false status
```
/opt1/arduino/arduino-ide_2.0.0-beta.4-snapshot.51195eb_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli compile --fqbn STM32:stm32:GenF4 --clean --build-path . --format json /home/andrew/Arduino/sketch_mar28b{
  "compiler_out": "Couldn't deeply cache core build: Rel: can't make . relative to /tmp/arduino-core-cache\nRunning normal build of the core...\n",
  "compiler_err": "",
  "builder_result": {
    "build_path": "."
  },
  "success": false
}
```
and it successfully generates a **compile_commands.json**
```
[
 {
  "directory": "/tmp/arduino-language-server071191930",
  "arguments": [
   "/opt1/arduino/dotarduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-g++",
   "-mcpu=cortex-m4",
   "-mfpu=fpv4-sp-d16",
   "-mfloat-abi=hard",
   "-mthumb",
   "@./sketch/build_opt.h",
   "-c",
   "-Os",
   "-w",
   "-std=gnu++14",
   "-ffunction-sections",
   "-fdata-sections",
   "-nostdlib",
   "-fno-threadsafe-statics",
   "--param",
   "max-inline-insns-single=500",
   "-fno-rtti",
   "-fno-exceptions",
   "-fno-use-cxa-atexit",
   "-MMD",
   "-I/home/andrew/Arduino/sketch_mar28b",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/avr",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/LL",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/OpenAMP",
...
```

ag88, 2021-03-29:
the language server toolchain architecture apparently seem to be
arduino-ide2 -> vscode-arduino-tools (module) -> arduino-language-server -> clangd
in **inols-err.log**
```
--> initialize(file:///home/user/Arduino/sketch_mar28b)
running:  /opt1/arduino/arduino-ide_2.0.0-beta.4-snapshot.51195eb_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli compile --fqbn STM32:stm32:GenF4 --only-compilation-database --clean --source-override /tmp/638675773 --build-path /tmp/arduino-language-server071191930 --format json /home/user/Arduino/sketch_mar28b
```
hence compile_commands.json is actually generated by arduino-cli using the --only-compilation-database option

i went into /tmp/arduino-language-server071191930 and tried
```
/opt1/arduino/arduino-ide_2.0.0-beta.4-snapshot.51195eb_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli compile --fqbn STM32:stm32:GenF4 --only-compilation-database --clean --build-path . --format json /home/user/Arduino/sketch_mar28b
{
  "compiler_out": "",
  "compiler_err": "arm-none-eabi-g++: error: ./sketch/build_opt.h: No such file or directory\n",
  "builder_result": {
    "build_path": "."
  },
  "success": false
}
```
^^^ in running the above command i've removed the source override option and did the compilation in the /tmp/arduino-language-server634547158 directory

however if I remove --only-compilation-database option the build completes, though with a success : false status
```
/opt1/arduino/arduino-ide_2.0.0-beta.4-snapshot.51195eb_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli compile --fqbn STM32:stm32:GenF4 --clean --build-path . --format json /home/user/Arduino/sketch_mar28b{
  "compiler_out": "Couldn't deeply cache core build: Rel: can't make . relative to /tmp/arduino-core-cache\nRunning normal build of the core...\n",
  "compiler_err": "",
  "builder_result": {
    "build_path": "."
  },
  "success": false
}
```
and it successfully generates a **compile_commands.json**
```
[
 {
  "directory": "/tmp/arduino-language-server071191930",
  "arguments": [
   "/opt1/arduino/dotarduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-g++",
   "-mcpu=cortex-m4",
   "-mfpu=fpv4-sp-d16",
   "-mfloat-abi=hard",
   "-mthumb",
   "@./sketch/build_opt.h",
   "-c",
   "-Os",
   "-w",
   "-std=gnu++14",
   "-ffunction-sections",
   "-fdata-sections",
   "-nostdlib",
   "-fno-threadsafe-statics",
   "--param",
   "max-inline-insns-single=500",
   "-fno-rtti",
   "-fno-exceptions",
   "-fno-use-cxa-atexit",
   "-MMD",
   "-I/home/user/Arduino/sketch_mar28b",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/avr",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/LL",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb",
   "-I/opt1/arduino/dotarduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/OpenAMP",
...
```

utherbone, 2021-06-03:
**Additional info for this bug**
Similar to: https://github.com/arduino/arduino-ide/issues/291 and archived repo: https://github.com/arduino/arduino-pro-ide/issues/244

board to STM32
**Screenshots**
[image]

If change to board AVR nano shows this:
[image]

**Additional context**
I searched here did not find much about the new updates I hope have helped. sharing what happened

**Distribution**
 - OS: Ubuntu
 - Version: 20.04.2 LTS
**Kernel**
 - kernel: Linux
 - Version: 5.4.0-73-generic

closed 2021-11-29 by fstasi