arduino/arduino-ide #224: IDE 2.0 não faz upload

nivatec, 2021-03-15:
instalei a ide 2.0 mas não consigo fazer upload   sempre apresenta erro. por favor algem pode me ajudar

nivatec, 2021-03-15:
I installed ide 2.0 but I can't upload it always shows an error. please someone can help me

BelKed, 2021-03-15:
What error?

First you need to select board and pin. Then you can upload...

nivatec, 2021-03-15:
> What error?
> 
> First you need to select board and pin. Then you can upload...

the error you present is this:
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=6692, tid=0x0000000000000598
#
# JRE version: Java(TM) SE Runtime Environment (8.0_281-b09) (build 1.8.0_281-b09)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.281-b09 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows

ubidefeo, 2021-03-19:
@nivatec 
the error you report is about Java, the IDE 2.0 does not require Java.
Also please provide detailed information as specified in the issue template.

Operating system
ways to reproduce
versions and so on

thanks

closed 2021-09-08 by per1234