arduino/arduino-ide #227: IDE does not bring up local file links in sketches when using relative URLs.

bperrybap, 2021-03-15:
The IDE allows embedding URLs in a sketch which the user can click on that will bring up a browser for the link.
On the pre 2.0 IDE, the URL could be a local file reference using a relative URL.
The root of the relative path is the directory for the sketch.
The 2.0 IDE fails to bring up a browser with the file when using a relative path URL.

Here is an example of a relative URL from my hd44780 library in the example sketch "Documentation"
`
// hd44780 Library ReadMe: file://./docs/README.html`

Relative links for local files are very useful for providing the ability to deliver local documentation to the user that can be accessed from the sketch directly from the IDE by simply clicking on it.

I have not looked at the code in the IDE to see what changed from pre 2.0, but I would assume that this is an easy fix, perhaps something as simple as the IDE not setting the current working directory to the sketch directory prior to either looking for the file or before starting up the browser so the browser is looking in the wrong location.

Note that I have seen MANY people over the years mistakenly believe that the RFCs 8089, 1630, 1738,  don't allow or preclude the use of relative paths.
This is incorrect. When specifying a URI, the "file-auth" (hostname) field can be omitted.
When specifying a path beginning with dot "." then the file-auth field has been omitted and the URI is specifying a local relative path.
Also, there is a magic "file-auth" name of "localhost" that is supposed to be supported that basically means the same as omitting the "file-auth" field.
So in this specific case for the IDE:
i.e.
`file://./docs/README.html`
and
`file://localhost/docs/README.html`

should bring up the same file if the root is the directory where the sketch is.

As is, currently with IDE 2.0 neither of these local file access URLs works.
But the relative path URL does work with pre 2.0 IDE.
(There seems to be an issue when using "localhost" on pre 2.0 IDE)

IMO, this would be a good thing to fix in IDE 2.0, work with relative paths for local files so that sketches can continue to have the capability to bring up locally provided files for things like documentation.
Without this capability, it is difficult for users to locate the files and bring them up so the experience is not nearly as nice as what they get with pre 2.0 IDE.

There is a bit of initial discussion in this Arduino thread:
https://forum.arduino.cc/index.php?topic=731043.0

But as I said earlier, the RFCs do not preclude using local relative paths and the do actually define how local relative paths should work.
What the RFCs don't specify is the root location for relative paths as that is implementation specific.
In the case of the Arduino IDE, to make things work for all sketches, it is best if the root is the directory of the sketch.

bperrybap, 2021-03-23:
Attached is a zip file that can be used to demonstrate the issue and also help verify that the issue has been fixed.
[Documentation.zip](https://github.com/arduino/arduino-ide/files/6190319/Documentation.zip)
To use:
- unzip the zip image into the sketchbook area and it will create a directory and sketch called "Documentation".
- Bring up the "Documentation" sketch.
- click on the links to test

There are several links in the sketch.
All of them work with pre 2.0 IDE, but the one on line 13 that references a local file:
`// hd44780 Library ReadMe: file://./docs/README.html`
does not work with IDE 2.0

ubidefeo, 2021-03-23:
thank you for providing this example, @bperrybap

bperrybap, 2021-03-23:
I looked around but couldn't seem to locate the area in the tree where the code that handles this resides.
Could you provide me some sort of general pointer to get me to at least the proper directory of where to look for the code that handles this lives?

ubidefeo, 2021-03-23:
I think this is a feature of the Monaco editor, we might not have much control over it but it can be investigated

bperrybap, 2021-03-23:
I'll dig a bit deeper with some profiling tools. It looks like it is trying to locate/open a file, but can't find it.
I need to get some more details to see exactly what it is doing but my suspicion is that the current working directory is in the wrong location so it can't find the file from the relative path.

bperrybap, 2022-03-05:
I just tried IDE 2.0.0 RC4 and it seems to be able to locate and handle local files referenced using relative URLs.
However, the IDE is mishandling local file when it is an  html file.
So this should change from being an enhancement to a BUG.

The IDE handles local file URLs by pulliing the file into the IDE.
But when the file is an html file the IDE treats the file as if it were plain text so it isn't really usable.
If the IDE cannot properly render html files, the IDE  should fire up a browser and hand the URL of the local file to the browser so the browser can handle the local html file just like what happens with http or https links.

bperrybap, 2022-07-25:
Ok and update on this. It kind of sort of works now but still has an issue for html files.
The IDE has appears to have been fixed to be able to properly locate local files referenced by relative URL reference, but there is an issue related to files that contain html.
If a sketch has a url to a local file like:
file://./docs/README.html
The IDE now brings up that file as if it were a text file into the IDE viewing window.
i.e. it shows the raw html which is not very useful.
The IDE needs to interpret the html itself, or even better just fire up a browser and pass the browser the full path to the local file and let the user's browser do the html rendering.

It would be nice if the code that handles this could check the file extension and if it is .htm or .html then fire up a browser to handle the file rather than bringing up the raw html into the IDE which is not helpful.