arduino/arduino-ide #228: Add the 'Search all Sketch Tabs" checkbox to the Find dialog

esp32croft, 2021-03-15:
The old IDE allowed you to search across all tabs of a sketch (and replace). This useful checkbox option has been lost in v2 of the IDE so you have to search each tab in turn.

kittaakos, 2021-03-16:
Thanks for reporting it, @esp32croft.

We added the _Search_ view to the IDE: https://github.com/arduino/arduino-ide/pull/202. The new view supports find, replace, exclude/include globs, and compare editor if you want to double-check your changes before replacing the content. It works on all tabs.

You can try it out from the nightly builds if you want to: https://github.com/arduino/arduino-ide#nightly-builds.
We would love to hear your feedback. Thanks!

esp32croft, 2021-03-16:
Hi @kittaakos,
I've just tried the nightly build but didn't find a way of searching all the tabs for the selected text. The search still only finds matches in the selected tab. I've included a screenshot which shows what I'm getting;
[image]
 I was expecting an option to search across all tabs from this dialog. I'm wondering if there is a setting somewhere that I've not enabled.

ubidefeo, 2021-03-16:
@esp32croft 
you are still using the "find" command, not the search feature.
"Find" searches in the current tab, the "search" feature is on the left, in the sidebar

kittaakos, 2021-03-17:
> The search still only finds matches in the selected tab. I've included a screenshot which shows what I'm getting;

You're very close, @esp32croft. If you want to search in all tabs, you have to use the new _Search_ view.

[image]

See the screenshot:

[image]

esp32croft, 2021-03-17:
Thanks @kittaakos , @ubidefeo ,
I didn't spot it was a NEW search feature rather than an extension to Find as previous. I blame muscle memory and should RTFM.
Would be nice if the Search could pick up the highlighted text in the current tab to pre-fill the Search, as it does for Find.

ubidefeo, 2021-03-18:
hi @esp32croft 
no worries, new features are often not immediately noticeable :)

> Would be nice if the Search could pick up the highlighted text in the current tab to pre-fill the Search, as it does for Find.

the editor find is its own component part of the Monaco editor.
I guess for this kind of stuff you can build up some muscle memory for copy paste 😀 

We won't implement this, since the search facility is a "search within workspace" and its functionalities go beyond searching for text. Pre-filling it would require moving text from one context to another.
We'd like to save development time for more pressing issues

thank you for your contribution 🙏🏼

closed 2021-03-18 by ubidefeo