arduino/arduino-ide #229: Display error messages from Compiler and Linker in red...

trlafleur, 2021-03-16:
OSX 11.1
2.0.0.b3

Older IDE reported errors in red to make it easy to spot in compiler and linker output

kittaakos, 2021-03-16:
This is a great idea. We used to have this feature but it was not working correctly so I removed it. Now, I quickly added it back to my local env as it was a couple of lines of code. It's not what you would expect 😊 This is how it looks like:

[image]

Unfortunately, the CLI  (that powers the new IDE) doesn't use the streams correctly. (See the CLI API [here](https://github.com/arduino/arduino-cli/blob/414ab001a235c213f96e00f185f082783118e109/rpc/commands/compile.proto#L50-L51)). Only the the `Compilation error: Error: 2 UNKNOWN: exit status 1` part goes to the error stream, so the IDE cannot do much and leaves the rest white. We need some help from the CLI team with this issue.

Xtreemtec, 2021-04-05:
would be great if that will be turned on in B4 or B5 again..  Running b4 now and really searching for the errors as everything is white.  
Also would be great if you could click on the error and the IDE would focus / highlite on that error.

alve89, 2021-06-02:
I think the basic problematic is not solved yet (Beta 7):
[image]

Errors are still not "highlighted" or obviously visible - or is any configuration needed for this?

per1234, 2021-06-02:
>I think the basic problematic is not solved yet

Correct, as should be clear from the fact that this issue is still open. We close issues once they are resolved.

@alve89 for support requests, please use the dedicated category for Arduino IDE 2.x on the Arduino Forum:
https://forum.arduino.cc/c/software/arduino-ide-2-0-beta/93
I'll be glad to help you out over there.

alve89, 2021-06-02:
> Correct, as should be clear from the fact that this issue is still open. We close issues once they are resolved.
> 
My bad, I misinterpreted the "closed" icon! I thought it is for this issue but it is for #353, so my post can be ignored. Thank you for your link anyway!

per1234, 2021-11-04:
>would be great if you could click on the error and the IDE would focus / highlite on that error.

There is a dedicated issue tracking this request here: https://github.com/arduino/arduino-ide/issues/118

commit 3a17550a (2022-06-17): Reveal the error location after on failed verify.

Closes #608
Closes #229
commit 842a078b (2022-06-17): Reveal the error location after on failed verify.

Closes #608
Closes #229
commit 63f1d851 (2022-06-21): Reveal the error location after on failed verify.

Closes #608
Closes #229
commit d6f4096c (2022-06-21): Reveal the error location after on failed verify.

Closes #608
Closes #229
closed 2022-06-21 by kittaakos