arduino/arduino-ide #232: Unable to resize IDE, Unable to move Select Board Dialog and difficulty in Selecting Board for arduino ide 2.0 beta.

abnair1982, 2021-03-17:
**Describe the bug**

When I open Select Board Dialog, the OK button to confirm selection is hidden. I am unable to resize the IDE or move the Select Board Dialog. As a result I am unable to select board easily. I have a mac style dash. So I have to drag the ide to the right to make the button visible. I tried to move and resize Select Board Dialog. But I was not able to move the dialog. Please check the screenshot

**To Reproduce**
1. Open IDE
2. Click On 'Select Other Board & Port'
3. 'Select Board' Dialog will open. 'OK' button to confirm selection is hidden.
4. Try to drag or resize 'Select Board' dialog. It will not allow you to move the dialog
5. Try to resize the IDE. It will not allow to resize IDE.

**Expected behavior**
1) Ability to resize the IDE. 
2) Ability to move or resize the "Select Board"  Dialog opened in IDE
3) Have 'OK' button at place that will be always easily accessible.

**Screenshots**
On opening the select board dialog:
[image]

On moving the ide to right:
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: 18.04
- Arduino Ide Version: arduino-ide_2.0.0-beta.3_Linux_64bit

**Additional context**
I have to move the IDE to the right for the buttons to became visible.

kittaakos, 2021-03-18:
Thanks for letting us know, @abnair1982. I merged your bug report into an existing issue: https://github.com/arduino/arduino-ide/issues/96#issuecomment-801883221

closed 2021-03-18 by kittaakos

abnair1982, 2021-03-18:
> I merged your bug report into an existing issue: #96 (comment)

Thank you for your response.