arduino/arduino-ide #238: Deleting a sketch tab that declares a function used in other tabs doesn't report errors.

silvanocerza, 2021-03-18:
**Describe the bug**

Deleting a sketch tab that declares a function used in other tabs doesn't report errors.

**To Reproduce**

1. Create a new sketch
2. Save it
3. Select any board
4. Create a new tab named `tab.ino`
5. In the `tab.ino` declare a function `void foo() {}`
6. Save `tab.ino`
7. Call `foo()` inside `loop()` in the main sketch file
8. From the file explorer delete `tab.ino`
9. Verify the `foo()` call inside `loop()` is not underlined as error
10. Type `foo` inside `loop()` and verify the `foo()` function is shown in the autocomplete windows

**Expected behavior**
I'd expect `foo()` to be underlined as error after `tab.ino` is deleted and not to be shown in the autocomplete window anymore.

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: 2.0.0-beta.4-nightly.20210318
 - Date: 2021-03-18T03:10:26.441Z
 - CLI Version: 0.17.0 alpha [ab587583]

**Additional context**
None.

ubidefeo, 2021-03-18:
Well spotted, @silvanocerza 
This is very likely because the files it is analysing are still the temporarily built ones.
This means a rebuild is required at deletion of a file/tab

per1234, 2021-11-25:
> 8. From the file explorer delete `tab.ino`

The bug also occurs when you use the primary UI to delete `tab.ino` (click the downward pointing triangle button on the right side of the tab bar, then "**Delete**").

per1234, 2022-03-31:
Closing as fixed by https://github.com/arduino/arduino-language-server/commit/3b7271afba1eaaaf2e56a545cc914e5a9a37381f
Thanks Cristian!

[image]

closed 2022-03-31 by per1234