arduino/arduino-ide #239: How can we add an extension to Arduino IDE 2.0?

raspiduino, 2021-03-19:
**Is your feature request related to a problem? Please describe.**
I always want to try adding an extension to Arduino IDE

**Describe the solution you'd like**
Somehow to add an extension to the Editor.

**Describe alternatives you've considered**
None

**Additional context**
What can the extension be written in? Javascript or any other languages?
If this feature is available now, can you add an example?

ubidefeo, 2021-03-20:
@raspiduino 
we have a way of supporting VS Code extensions but we will not support the ones not developed by us.
If you develop an extension and it causes the user to break the IDE or lose work we cannot be responsible for it.
This is the main reason why we do not advertise this feature.

You can look at documentation for Theia and develop extensions in TypeScript

closed 2021-03-20 by ubidefeo

raspiduino, 2021-03-20:
Thanks anyway!