arduino/arduino-ide #240: How to debug for Arduino Uno?

raspiduino, 2021-03-19:
**Is your feature request related to a problem? Please describe.**
In the blogpost about Arduino IDE 2.0 it said

> As of today, the debugger supports all the Arduino boards based on the SAMD and Mbed platforms (MKR family, Nano 33 IoT, Nano 33 BLE, Portenta, Zero). Maintainers of Arduino cores for third-party boards can add support for debugging by adding the relevant configuration parameters; a technical guide for this is coming.

So how can use debug for Arduino Uno? When will `a technical guide for this` come?

**Describe the solution you'd like**
Somehow to debug for Arduino Uno (and other boards)?

**Describe alternatives you've considered**
Use AVR Studio, but it's much heavier than Arduino IDE!

**Additional context**
Thanks for any help!

ubidefeo, 2021-03-19:
@raspiduino 
please search the issues before opening a new one.
There is a discussion in #87 regarding this.
One user has been able to do it, but it requires quite a lot of work and you'll need some custom made tools as well as setting fuses on the AVR.

You cannot simply debug a microcontroller.
I doubt we'll add official support for this, as the request is not wide and debugging an AVR can turn out to be complicated.

Please follow the tutorial posted by the user who successfully did it.

Thanks
ubi

tpimh, 2022-06-07:
DebugWire is one option, another option is a software debugger: [avr_debug](https://github.com/jdolinay/avr_debug).

closed 2022-12-02 by per1234