arduino/arduino-ide #242: Partial crash: Cannot save file or compile

quilkin, 2021-03-19:
**Describe the bug**
Semi-crash: 
- Bottom ppanel has message 'offlne' when it isn't (there is UNO on COM6)
- Choosing 'Verify/Complie' does nothing
- If i try to save,  get a message 'culd not save...' and asks me to save into a favourite editor and restart.

So, all crashed except for the editor itself ( I can kepp on editing)

**To Reproduce**
Steps to reproduce the behavior:
Unknown; I have been editing for about an hour without compiling

**Expected behavior**
These functions should work!

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.0.0 beta 3

**Additional context**
Add any other context about the problem here.

ubidefeo, 2021-03-19:
Hi @quilkin 
if the bottom status bar tells you that it's `offline` it means that the CLI has not managed to start correctly.
We have seen this in case another application uses the CLI daemon port 50051 or if some weird combination of firewall/antivirus is in place.

could you please try and temporarily disable any antivirus and just relaunch the IDE?
let us know if this solves your issue

quilkin, 2021-03-20:
This only happened once (after editing for about an hour) and it hasn't happened again (yet) , so it's not going to be productive to try disabling things. I wasn't using any other apps on the PC at the same time, except for a browser. The only anti-virus in use is Window's built-in one.
This may help: sometimes, when hitting 'upload', a brief yellow message flashes up in the output window, saying that the device is offline. This isn't the same message as I saw when the IDE crashed. This lasts for 2-3 seconds before the IDE proceeds as it should. 
For other reasons (notably the inadequacies of the slow serial output, noted elsewhere) I am not intending to use this IDE any longer, and will switch to the VisualMicro add-in for VS2019 instead.

ubidefeo, 2021-03-20:
thank you for letting us know
hope VisualMicro serves you well :)

closed 2021-03-20 by ubidefeo