arduino/arduino-ide #244: Empty entries in Tools -> Board menu and Board options missing in the menu

HilmarSt, 2021-03-19:
**Describe the bug**
I've installed several additional boards in IDE 1.8 - when I open the IDE 2.0, the Tools -> Board menu is partly empty and the Board options are not available.

**To Reproduce**
Go to the Tools -> Board menu; the problem is most probably only visible with a  configuration similar to mine.

**Expected behavior**
The IDE 2.0 should show (at least) the same entries as the IDE 1.8 .

**Screenshots**
[image]

[image]

**Desktop **
 - OS: Windows
 - Version: 10 / 64 Bit

mrbbp, 2021-03-19:
[image]

idem on macos 10.14.6, the title does not appear.
In my case only Digispark is blank but sub menu is readable...

[image]

mrbbp, 2021-03-19:
same prob on macos version too (entries #245, sorry for twice entry)

ubidefeo, 2021-03-20:
@mrbbp 
I have an idea of why this happens and it's related to cores installed via the Java IDE or CLI and not "tracked" by IDE 2.
Could you run this test for us?

Quit IDE 1.8.13 if open. In IDE 2.0 
* check wether you have MiniCore Additional URL
* * if you don't add it and restart the IDE
* * check if behaviour has changed

if the above didn't work
* uninstall MiniCore
* install MiniCore 

Let us know what works :)

HilmarSt, 2021-03-20:
@ubidefeo
I've done what you've proposed:
In IDE v1.8 I've 4 additional URLs, in IDE v2.0 none - so I added the URL for MiniCore: This does not change anything.
Then I uninstalled MiniCore and installed it again - that solves (a part of) the problem, (only) the entries for MiniCore a correctly visible again.
This is anyhow not very satisfying, because I have to do this manually for all additional URLs and the problem is, that there is no mapping between the URL and the name of the board package - that means that I can not see easily which of the many entries in the boards manager I have to re-install.

Beside this, a BIG compliment to the whole development team! You are doing a very good job, the turn-around time is much better compared to the old IDE.

HilmarSt, 2021-03-20:
Sorry, **I did not restart** the IDE after adding the URL.
Now I've added all URLs to the IDE v2.0, **restartet** the IDE and **everything looks ok.**
That means, no re-install is necessary.
The only remaining whish is, that the IDE v2.0 should import all additional URLs from the old IDE.

ubidefeo, 2021-03-20:
hi Hilmar

Happy to hear that worked.
aunfortunately the additional URLs in the Java IDE are stored inside the IDE's preferences together with a lot more stuff.
It is way safer if the user does a copy/paste.

We'll write some knowledge-base article about it, but thanks for bringing this up :)

ubi de feo

mrbbp, 2021-03-20:
@ubidefeo
Hello, i tried to reinstall my board (not the mini core)
it does not work..

But
i reinstall Beta3 (cause. what i. have to do was enough and fonctionnal) and launch it, choose my board, variant and upload method.
quit beta3
launch beta4
all the menu is good
But if i change the board, variant and upload menu disappeared again...

i redo, the b3 board config quit and launch b4... it's good again
so bug is reiterable

mrbbp, 2021-03-20:
back to b3 ;)